theoctal/octalbonescript #13: Problem using some IO pins on the beagle

PLorand, 2015-01-07:
Hello,

I think this is more of a question than an issue. So, I'm trying to use pins P8_38 and P8_39 - basically what I want to achieve is to use all 65 IO ports available on the beaglebone, but quite frankly it seems I'm quite unable to do so.

I have disabled eMMC (booting from microSD), also disabled the two HDMI outputs that use up some of the IO ports, so the capemanager looks something like this when using ocalbonescript to manage the pins:
 0: 54:PF---
 1: 55:PF---
 2: 56:PF---
 3: 57:PF---
 4: ff:P-O-- Bone-LT-eMMC-2G,00A0,Texas Instrument,BB-BONE-EMMC-2G
 5: ff:P-O-- Bone-Black-HDMI,00A0,Texas Instrument,BB-BONELT-HDMI
 6: ff:P-O-- Bone-Black-HDMIN,00A0,Texas Instrument,BB-BONELT-HDMIN
 7: ff:P-O-L Override Board Name,00A0,Override Manuf,OBS_UNIV
 8: ff:P-O-L Override Board Name,00A0,Override Manuf,OBS_AUDIO
 9: ff:P-O-L Override Board Name,00A0,Override Manuf,OBS_HDMI
10: ff:P-O-L Override Board Name,00A0,Override Manuf,cape-bone-iio

I'm running on a Beaglebone Rev-B (also have Rev-C, but I don't think that is relevant), and Debian 3.8.13-bone50, NodeJS v0.10.33.

Do you happen to know a solution how I could free up all 65 IO ports and actually be able to control them?.. Please note, that the ports that don't have anything else assigned work without any issues. I also checked for conflicts, but none are reported. Really, at the moment I have used up all my ideas about achieving my goal, which is very important as it involves using the beagle in a project I'm working on.

I will highly appreciate any help you can offer in this regard.

Thanks,
Lorand.

adityapatadia, 2015-01-07:
Actually, I have also never been able to use all the IO pins of BB. As per my understanding, the pins used by eMMC on board can't be free so we can't use those pins for IO. Those are P8_03 to P8_07. Frankly, when I disable eMMC just the way you did, I am not able to boot the board so I also can't use those pins used by memory cad reader. 

If you come across some solution, please post here, we are in the same boat :)

PLorand, 2015-01-07:
So basically what's the maximum number of pins you've managed to use on the beagle?..

Why should eMMC remain active when you are booting from microSD? That doesn't sound too logical to me...

adityapatadia, 2015-01-07:
I am not booting from MicroSD so I think that's the reason I can't disable eMMC. Correct me if I am wrong. The Maximum I get is 54 pins. I cant use P8_03 to P8_07 and P8_20 to P8_25.

PLorand, 2015-01-07:
I might say something very stupid right now, but within octalbonescript the pin number used (based on: http://beagleboard.org/static/images/cape-headers-digital.png ) is the one in yellow (in the center), or the side ones?... Because I'm referring to GPIO_38 and I think you're talking about the pin numbers from the center... :)

adityapatadia, 2015-01-07:
On OctalBoneScript (OBS) the pin numbers are the one in yellow. The one you are saying GPIO_38 is P8_03. It's confusing at times, hope I helped :)

PLorand, 2015-01-07:
Yeah definitely :) Let me try something... I tried using P8_38, while measuring P8_03, let me see if I set up pins 03 and 04 and look around again.

adityapatadia, 2015-01-07:
Just published new version. Just run ```npm update``` once before you proceed.

PLorand, 2015-01-07:
Hmm, if I change P8_38 to P8_03 and P8_39 to P8_04, I get the following error when trying to set up the pins: TypeError: Cannot read property 'gpio' of undefined

Any ideas?..

adityapatadia, 2015-01-07:
Hope the upgrade solved your problem.

PLorand, 2015-01-07:
Unfortunately still upgrading because I have several modules installed at the moment - so it takes a while...

PLorand, 2015-01-07:
The npm update screwed up some things, I have to restart it... hopefully if I solve what it was complaining about it will work again... :(

PLorand, 2015-01-07:
Okay, I've finally managed to more or less complete the update. Now I get the following error:
error: onComplete problem: Error: ENOENT, open 'undefined/state'
error: Unable to configure mux for pin [object Object]: Error: ENOENT, open 'undefined/state'
error: onComplete problem: Error: ENOENT, open 'undefined/state'
error: Unable to configure mux for pin [object Object]: Error: ENOENT, open 'undefined/state'

/usr/local/lib/node_modules/octalbonescript/src/hw_universal.js:77
            callback(state.trim());
                           ^
TypeError: Cannot call method 'trim' of undefined
    at /usr/local/lib/node_modules/octalbonescript/src/hw_universal.js:77:28
    at fs.js:207:20
    at Object.oncomplete (fs.js:107:15)

Is this from youor from me?...

adityapatadia, 2015-01-08:
The error message essentially means that the pin is being used by some other cape and it is not available to OBS.

PLorand, 2015-01-08:
So I did some further testing today, and it seems I can use 53 GPIO pins, in other words, all the GPIOs besides the eMMC reserved ones (3-6 and 20-25). What I don't understand is why can't I use those eMMC pins, when eMMC is not even used on the board because I'm using uSD for booting... Any ideas?..

Another question that came up to me. Can you use all GPIO pins with interrupt input, or just some? I would like to have rising edge, or high interrupt on some GPIO inputs. Is this possible?

PLorand, 2015-01-08:
Okay, I've continued looking into this issue and it seems that it's kind of a kernel "fault", because it tries to load the bootloader from the eMMC, even before it would load the DT (to disable the eMMC). There are some workarounds to handling this which include creating a new kernel which disables this eMMC booting, but I think that's a bit too much of a hustle just to get 12 pins working (although for me at the moment would be great to have).

If you are interested in some stuff I found you could read about it here:
https://github.com/modmaker/BeBoPr/wiki/BeBoPr-goes-black
https://groups.google.com/forum/#!topic/beagleboard/pSOre-xZz8o

I have an urgent project going on the Bbone at the moment and I really don't have a lot of time to experiment building custom kernels and reconfiguring everything (especially when I don't have a lot of experience doing this), so probably for outputs I will use some shift registers to create those additional pins needed for me. Once I finish this, I might try and experiment a little more with these workarounds - but I don't find them to suitable in a production environment when you want to deploy multiple boards...

Anyways, do you happen to have some info on using the IO pins as inputs with interrupts? Can an interrupt be set up on any input pin? How can I set up interrupts with OBS?...

adityapatadia, 2015-01-08:
Hey, I also thought the same that editing Kernel is not a good idea for a few pins. 
For interrupt, you may try it on BB and it should work. Frankly, I have not made any changes to interrupt code after I forked from Bonescript. I request you to refer to attachInterrupt function of bonescript.

If you face any error, let me know I will correct in OBS.

PLorand, 2015-01-08:
Okay, I will test and let you know if any issues occurred. Do you know anything about restrictions to how many gpio (or which) can have interrupts attached?..

adityapatadia, 2015-01-08:
Sorry no idea. You can check these files for reference of modes of each pin. https://www.dropbox.com/s/oslyh08f5dk66ks/BeagleboneBlackP8HeaderTable.pdf?dl=0
https://www.dropbox.com/s/zespd5hgquq4dac/BeagleboneBlackP9HeaderTable.pdf?dl=0

PLorand, 2015-01-08:
Thanks, I've been using these tables as well. :)

So, I tried using interrupts and for each pin I try attaching an interrupt, I get the following error: "error: attachInterrupt: pin P8_07 not already configured as GPIO". Does this say anything to you?.. I tried googling the error, but I can't seem to find anything relevant in regards to it. pinMode is set to INPUT_PULLUP.

The implementation looks something like this:
var b = require('octalbonescript');
b.pinMode("P8_07", b.INPUT_PULLUP);
b.attachInterrupt("P8_07", true, b.CHANGE, interruptCallback);

Also, in the interruptCallback I've printed whatever gets transmitted to it in JSON format, and this is the following: 
{"pin":{"name":"TIMER4","gpio":66,"mux":"gpmc_advn_ale","eeprom":41,"key":"P8_07","muxRegOffset":"0x090","options":["gpmc_advn_ale","NA","NA","NA","NA","NA","NA","mmc1_sdcd"]},"attached":false,"err":"attachInterrupt: pin P8_07 not already configured as GPIO","configured":false}

adityapatadia, 2015-01-08:
The pinMode function is fully async so you should use callback for attaching interrupt. Try:
```javascript

var b = require('octalbonescript');
b.pinMode("P8_07", b.INPUT_PULLUP,function(resp,pin){
    b.attachInterrupt(pin, true, b.CHANGE, function(resp){
        console.log(resp);
    }); 
});

```

PLorand, 2015-01-08:
Okay, great! Interrupts are working well. :) Thanks for your help!

Internal pullups are also working. I've applied them to all the working pins and randomly tested them and they all seem to be working well. So I can confirm that you can have interrupts assigned to all pins that you can work with (65 x io - 2 x i2c - 10 x eMMC = 53 working gpios).

Would have been really great to get completely rid of the eMMC and be able to use those pins, but recompiling the kernel to do just that seems a bit steep for me. :(

adityapatadia, 2015-01-08:
Hey, thanks for testing it :) Hope you loved OBS. Let me also know if you find anything for eMMC.

closed 2015-01-08 by adityapatadia