theoctal/octalbonescript #14: merge fork

boneskull, 2015-01-22:
Hi,

It would be helpful for BeagleBone users if you could work with @jadonk to get features and any fixes merged back into [bonescript](/jadonk/bonescript).

As mentioned in jadonk/bonescript#83, the changes made here are mix of style and functional changes, which make straight PRs difficult.

If I may offer some observations:

1.  A handful of packages depend on [bonescript](/jadonk/bonescript).  No packages depend on **octalbonescript**.  
2.  While bug fixes may have been made in some areas, various incompatibilities and other bugs have been introduced in this fork (as per julianduque/beaglebone-io#17).  These may be few, but because of this, packages dependent upon **bonescript** will find it difficult to justify switching.
3.  **bonescript** seems to be the "officially sanctioned" package, as visiting the web site of your BBB will describe its API.
4.  **octalbonescript** has more activity in the past year than **bonescript**.  
5.  Re-forking **bonescript** and manually applying functional changes and fixes may be the best way to do this.
6.  Users do not care about the code style of these two projects; simply whether or not they work.

And an opinion:

- It may be appropriate to add @adityapatadia as a maintainer of **bonescript**, since he has the time and inclination.  When somebody is improving my project more than I am, I'm happy to let them do it.

Normally, I'd offer my axe, but I'm a bit overburdened at the moment. 

TL;DR: Please merge the fixes and features back in, one way or another.

Thanks!

adityapatadia, 2015-01-22:
Check my answer in https://github.com/julianduque/beaglebone-io/issues/17

adityapatadia, 2015-01-22:
You have been not so polite with me on this page https://github.com/julianduque/beaglebone-io/issues/17 but that is fine. 

https://github.com/jadonk/bonescript/issues/100 check this issue. I suggested some feature, and no one is ready to reply to that. I reported this https://github.com/jadonk/bonescript/issues/82 and when it was approved, I sent https://github.com/jadonk/bonescript/pull/83. That one was closed without much concern. That has obviously brought us some disappointment.

We had basic requirement that we have a stable library at that point of time. We found it easier to fork and maintain on our own rather than wait for PRs to merge. Some of the changes we made were drastic in nature like making octalbonescript fully asynchronous and these kind of decisions would have taken a long time to decide by @jadonk. https://github.com/jadonk/bonescript/issues/75 is 11 months old issue and no work has been done on the same. @jadonk himself agrees that many changes in octalbonescript are such that they can't be merged in bonescript. On other hand those changes are transparent to users and we strongly believe that there is need for some drastic changes. We have for example, made octalbonescript ready for 3.14 kernel which will be released to BBB users in future.

We don't have any marketing purpose in mind while we develop this. The community should applause the work and efforts put by us. On the other hand, Github is full of examples where one stalled project is forked by other people and being maintained in nice manner then the original author can do. One prominent example I can think of is Python Pillow library. We also took care while naming that the fork and the original libraries are clearly distinguished by users. Please don't blame us for improving a stalled project. 

As for the website part, there is no wrong word written on there. Octalbonescript IS most reliable library for BBB . There are lots of users out there including you who are using octalbonescript rather than official bonescript and the reason is not marketing but the reason is we put our hard work to make it usable.

I would like to know views of @jadonk on this matter and how he is planning to develop bonescript in future. As I said, I can merge back to bonescript if there is agreement about the coding style, functionality and under the hood structure of library. Let's give the community a good library, be it octalbonescript or bonescript.

boneskull, 2015-01-22:
Let me apologize for sounding impolite--that was not my intent.  Sometimes I use metaphors when I probably should not.  Understand that this is not a personal attack--I'm simply looking at the history, weighing it against what you claimed, and coming to different conclusion.  

> We don't have any marketing purpose in mind while we develop this. The community should applause the work and efforts put by us.

**octalbonescript** is good code to be sure, but expecting adulation from a community for your efforts is a bit misguided.

> Please don't blame us for improving a stalled project.

I prefer this library over **bonescript**.  It *is* improved, in my opinion; that's not my complaint.  I believe you've done good work here, and that's to be commended.

But, each one of your changesets could have been a PR, and merging could happen more deliberately--but that doesn't mean you can't use your own fork while you wait & hash out changes.  Refactors could have been issues and discussed, instead of unilateral decisions.  Instead, we have two very different codebases which will be a pain to integrate.

Make no mistake; I see a failure on *both* ends here.  If it sounds like I'm calling you out, it's because you actually responded.  I don't see any comment here from @jadonk, which is understandable for someone who either has no time or interest in a project.  You can't blame somebody for losing interest or having other priorities, however--I have a bunch of projects just collecting dust--ideas that were good at the time, but then I got another idea... :smile: 

> As for the website part, there is no wrong word written on there. Octalbonescript IS most reliable library for BBB .

Perhaps anecdotally, but this remains an opinion unless you have data to prove it.  

> Let's give the community a good library, be it octalbonescript or bonescript.

100% agreed. If @jadonk wishes to abandon **bonescript**, that's fine too!

adityapatadia, 2015-01-23:
Thanks for understanding. Let's hear @jadonk's views on this. If he wants to be occasional contributor and suggest ideas for octalbonescript, I am more than happy to add him as maintainer. If he is ready to add me as maintainer of bonescript and we agree to make changes after a deliberation, I am also happy to accept invitation.

Meanwhile, I am going to continue development of octalbonescript.

Cheers!!

hardikpandya, 2015-01-24:
Hey Chris (@boneskull), Thanks for writing in. I see that you took time to bring the discussion forward that should have been between us (I'm the second guy in the "we" that @adityapatadia mentions) and @jadonk a long time ago.

We started developing projects with the BeagleBone almost 20 months ago now. Back then there was only bonescript. Our needs were simple, to have analog and digital functions to work reliably every time and in a fast manner. Bonescript DID prove a great starting point and it was a reason we started using BeagleBone in the first place.

But we quickly (within days of our initial testing with some basic code) ran into almost inexplicable errors and roadblocks in our tests.

The thing with open-source software is (and I'm sure you know this better than us) that you are pretty much on your own for the most part, that is - understanding the source of the thing you are using (for us, it was Bonescript), trying to engineer your makeshift solutions to have your thing work, etc. We spent nights and days asking questions on Twitter to @jadonk and posting issues on Github.

Creating Octalbonescript was in no way meant to harm @jadonk's or anyone'e reputation nor was it to steal someone's work. We wanted to have a code working for US and we didn't really want to "sell" or profit from it. That was the reason it's openly available to view/fork/comment on, even today. That's the reason for putting those strongly opinionated words on the website as well because this fork of bonescript DID work reliably enough for us and for those functions we mentioned.

Hence, for the beginners and early-testers, this code is way more suited for initial testing, with the examples mentioned on the TI and BeagleBoard.org website. And that's the claim we stand by even today.

We would be glad if @jadonk happily admitted in past that he would not be able to quicken the development of bonescript (which is perfectly fine of course) but our repeated approaches were met with long delays and more frustration on our end since our projects kept getting delayed while we waited for answers from the maintainer of bonescript.

We had to move on. And this is our way of moving on. The pace and direction of development wasn't right for us, we made our own flavour. No disrespect to @jadonk in any way of course. 

Also, thank you again for your comments, we love that people like our work so far and have been using Octalbonescript in their developments. The creator of bonescript STILL of course is very much @jadonk but I'm afraid Octalbonescript and bonescript have come a long way away from each other in a lot of ways to actually consider merging and refactoring the whole thing to work as one - and it would also mean investing a lot of time on both, ours and @jadonk's end.

Cheers!

framboosejh, 2015-01-24:
Hello,
I'm just an enduser and i'm learning to program the BeagleBone Black. I first just tried the path of Beagleboard.org with javascript and BoneScript because i liked the idea of having a webbased IDE like cloud9. but i ran in a lot of issues using bonescript even with capes like the bacon cape which should work out of the box, and yes it works with the test.sh in the Linux shell but not with bonescript, despite the fact that there are examples in Cloud9 and in the webserver from the BBB. than i decided to use C and C++ from the book of Derek Molloy. But that is een new programming language for me, so i have to learn that first. After that I found octalbonescript and gave that a try, because you said it is better and more stable, but immediately i ran in trouble again. After searching in the github i did understand that octalbonescript doesn't support Cloud9 and iosocket anymore.So i can't use it with a webbrowser. And it is hard enough to learn how Device Trees Overlays work. I think i have to go back to my Raspberry Pi because in their community they help each other and you see libraries which are working and even stand side by side like GPIO.rpi and wiringpi. I am disappointed in the BBB and his developers and community.

adityapatadia, 2015-01-24:
You can actually use cloud 9 IDE and just replace ```require('bonescript')``` with ```require('octalbonescript')``` and your code should work fine. Just read the readme once. If you face issues, you can report here and we will try to address them asap.

Thanks.

framboosejh, 2015-01-25:
Problem with octalbonescript and USR0.
I am trying to use octalbonescript in the following very simple program:
var b = require('octalbonescript');

b.pinMode("USR0", b.OUTPUT);
b.digitalWrite("USR0", 0);
It is supposed to put the hartbeat LED off, and it works. But when I want to put that LED on, so that it stays on  it doesn't work. Then program is then:
var b = require('octalbonescript');

b.pinMode("USR0", b.OUTPUT);
b.digitalWrite("USR0", 1);
When i try this with 'bonescript' it works, the LED goes on and stays on.
The program to do this comes from the book 'Programming the BeagleBone Black' from Simon Monk. When i use 'bonescript' there no extra overlays loaded:
root@beaglebone:~# cat $SLOTS
 0: 54:PF--- 
 1: 55:PF--- 
 2: 56:PF--- 
 3: 57:PF--- 
 4: ff:P-O-L Bone-LT-eMMC-2G,00A0,Texas Instrument,BB-BONE-EMMC-2G
 5: ff:P-O-L Bone-Black-HDMI,00A0,Texas Instrument,BB-BONELT-HDMI

When i use 'octalbonescript' there are two extra overlays loaded:
root@beaglebone:~# cat $SLOTS
 0: 54:PF--- 
 1: 55:PF--- 
 2: 56:PF--- 
 3: 57:PF--- 
 4: ff:P-O-L Bone-LT-eMMC-2G,00A0,Texas Instrument,BB-BONE-EMMC-2G
 5: ff:P-O-L Bone-Black-HDMI,00A0,Texas Instrument,BB-BONELT-HDMI
 7: ff:P-O-L Override Board Name,00A0,Override Manuf,OBS_UNIV
 8: ff:P-O-L Override Board Name,00A0,Override Manuf,cape-bone-iio

When i first put USR0 off with 'octalbonescript', which works fine and  put it on again with 'octalbonescript' this wil not work, but when i use 'bonescript' right after that, it will put USR0 on.

When i reboot the BBB and than put the LED USR0 with 'octalbonescript' on, that works fine and i can put it of again with 'octalbonescript' , but when i try to put it on again with 'octalbonescript' that wouldn't work. I just flickr's very short on and stays than of.

framboosejh, 2015-01-25:
I was looking futheron for this behavior and came on issue possible compatibility errors with P9_14 #8. When i use blinkled.js that program works fine.

closed 2015-02-15 by adityapatadia