theoctal/octalbonescript #15: USR LEDs don't work

framboosejh, 2015-01-25:


framboosejh, 2015-01-25:
I asked in issue #14 a question about de USR LEDs, i am new to github and don't know a lot about the use of it. By typing my question i realized that issue #14 was not the right place for my question and i decided to start with a new issue, this one. I will ask my question here again but that takes some time. I have first to check what i did over there and ask my question again. i wil do that in the coming days in the next comments.

adityapatadia, 2015-01-26:
Hey, glad that you opened new issue. I have already checked the comment on #14 and understood the problem. I will fix the same in a day or two. For the community, it would be great if you can put same issue here.

Thanks for reporting the bug.

framboosejh, 2015-01-26:
I wil give it a try..
Because the Blinkled.js program worked last time, i decided to simplify that program just blink USR0 and nothing else (program blinkusr0.js):
var b = require('octalbonescript')

var led = "USR0"

b.pinMode(led, b.OUTPUT);

var state = b.LOW;

b.digitalWrite(led, state);

setInterval(toggle, 1000);

function toggle() {
    if (state == b.LOW) state = b.HIGH;
    else state = b.LOW;
    b.digitalWrite(led, state);
}
This program works fine, USR0 is steady blinking. So i stopped it with Ctrl-C when  USR0 was off, USR0 stays off as aspected. Then i changed the program just to put USR0 on (program usr0on.js):
var b = require('octalbonescript')

var led = "USR0"

b.pinMode(led, b.OUTPUT);

var state = b.HIGH;
b.digitalWrite(led, state);

That works good to, USR0 goes on and stays on. Then i changed the program to put USR0 off (program usr0off.js):

var b = require('octalbonescript')

var led = "USR0"

b.pinMode(led, b.OUTPUT);

var state = b.LOW;
b.digitalWrite(led, state);
And yes, USR0 goes off  and stays off. To check everything i run the program to put USR0 on again. Now something unexpected happens. I just see a very short flash from USR0 and than it stays off!!
I don't understand. Is there something wrong with my program? I'm just learning JavaScript. So i tried blinkUSR0.js again, and that works good, stop it with Ctrl-C while USR0 is off, that works. And than put USR0 on again with USR0on.js. Yes, USR0 goes on and stays on.

I tried this several times and i get the same result even after a reboot to make sure that 'bonescript' isn't active.

adityapatadia, 2015-01-26:
Hey, I took a look at the problem. It is actually a problem in you program. OctalBoneScript is fully async library. That means that when you write ```pinMode``` function, its execution is not guaranteed to complete before the program goes to next line. That's why in your code, before execution of ```pinMode``` completes, ```digitalWrite``` function gets called. After that, when ```pinMode``` completes, it turns off the LED. That's the reason you see the small blink.

Below is the working code to overcome this situation:

```javascript

var b = require('octalbonescript');
var led = "USR0"

b.pinMode(led, b.OUTPUT,function(err,pin){
    var state = b.HIGH;
    b.digitalWrite(led, state);
});

``` 

So basically callback of pinMode function solves the issue.

closed 2015-01-26 by adityapatadia

framboosejh, 2015-01-27:
> 
> Hey, I took a look at the problem. It is actually a problem in you program. OctalBoneScript is fully async library. That means that when you write pinMode function, its execution is not guaranteed to complete before the program goes to next line. That's why in your code, before execution of pinMode completes, digitalWrite function gets called. After that, when pinMode completes, it turns off the LED. That's the reason you see the small blink.
> 
> Below is the working code to overcome this situation:
> 
> var b = require('octalbonescript');
> var led = "USR0"
> 
> b.pinMode(led, b.OUTPUT,function(err,pin){
>     var state = b.HIGH;
>     b.digitalWrite(led, state);
> });
> So basically callback of pinMode function solves the issue.
Thank you,
As you know it solves the issue indeed, i didn't understood the concept of async libraries and the influence on my program, so i have to look deeper in the async nature of javascript, node.js and the octalbonescript library.

with regards,

Evert Jan Henken=

framboosejh, 2015-01-27:
Hello, here i'm again,

I don't know if i may ask you a question this way, but i did try your code to overcome the situation i had and indeed it works.
But now i have another question:
I think i understand the callbackfunction and could right it this way to, isn't it right?

var b = require('octalbonescript');

var led = "USR0";

b.pinMode(led, b.OUTPUT,ledAan);

function ledAan() {
var state = b.HIGH;
b.digitalWrite(led,state);
}

I did test this and it works.

But I rather had a program that waited until pinMode completed and after that turn the LED on.
So i did try something by myself:

var b = require('octalbonescript');

var led = "USR0";

b.pinMode(led, b.OUTPUT,function(err,pin){While (!pin)});

var state = b.HIGH;
b.digitalWrite(led,state);

Of course it doesn't work, otherwise i wouldn't ask this question. I am not used to async programming, I am a old school (63 years old) basic programmer and in the early eighties i programmed some dBase software. So JavaScript, node.js and octalbonescript are new to me.

Regards,

Evert Jan Henken=

framboosejh, 2015-01-27:
And again,

I didn't realize that you can call my little peace of code, a program. It just put a LED on when you start it and nothing else. That's not really a useful awesome piece of work to show to my family.
Although the Blink LED program works and i understand some more about octalbonescript and the working of pinmode and async. I think i have to learn some more about pushbuttons and GPIO input and so on, so that i could at least put a led on and off with a button. I think that pinMode completes quicker than i can push a button. So i think my last mail about is not so relevant anymore.

I first have to practice a lot and maybe i have later on more sensible questions.

First off all i like octalbonescript much more now and certain your help, you are very friendly and helpful.

With regards,

Evert Jan Henken=

adityapatadia, 2015-01-31:
Hey, nice to know that things are working now. :)