theoctal/octalbonescript #16: gpio-related race condition?

boneskull, 2015-01-27:
Hi,

We can see that when my BBB boots, the GPIO is configured as this:

```
GPIOs 0-31, gpio:
 gpio-6   (mmc_cd              ) in  lo

GPIOs 32-63, gpio:
 gpio-52  (eMMC_RSTn           ) out lo
 gpio-53  (beaglebone:green:usr) out lo
 gpio-54  (beaglebone:green:usr) out lo
 gpio-55  (beaglebone:green:usr) out hi
 gpio-56  (beaglebone:green:usr) out lo
 gpio-59  (McASP Clock Enable P) out hi

GPIOs 64-95, gpio:

GPIOs 96-127, gpio:
```

When I run my script (this is johnny-five -> beaglebone-io fork -> octalbonescript), I get quite a bit of this output:

```
error: Writing to GPIO failed: Error: EPERM, operation not permitted
```

Then, examining the GPIO pins again results in:

```
GPIOs 0-31, gpio:
 gpio-2   (P9_22               ) in  hi
 gpio-3   (P9_21               ) in  hi
 gpio-4   (P9_18               ) in  hi
 gpio-5   (P9_17               ) in  hi
 gpio-6   (mmc_cd              ) in  lo
 gpio-7   (P9_42               ) in  lo
 gpio-14  (P9_26               ) in  hi
 gpio-15  (P9_24               ) in  hi
 gpio-20  (P9_41               ) in  lo
 gpio-22  (P8_19               ) in  lo
 gpio-23  (P8_13               ) in  lo
 gpio-26  (P8_14               ) in  lo
 gpio-27  (P8_17               ) in  lo
 gpio-30  (P9_11               ) in  hi
 gpio-31  (P9_13               ) in  hi

GPIOs 32-63, gpio:
 gpio-44  (P8_12               ) in  hi
 gpio-45  (P8_11               ) out lo
 gpio-46  (P8_16               ) in  lo
 gpio-47  (P8_15               ) in  lo
 gpio-48  (P9_15               ) in  hi
 gpio-49  (P9_23               ) in  lo
 gpio-50  (P9_14               ) out lo
 gpio-51  (P9_16               ) out lo
 gpio-52  (eMMC_RSTn           ) out lo
 gpio-53  (beaglebone:green:usr) out lo
 gpio-54  (beaglebone:green:usr) out lo
 gpio-55  (beaglebone:green:usr) out hi
 gpio-56  (beaglebone:green:usr) out lo
 gpio-59  (McASP Clock Enable P) out hi
 gpio-60  (P9_12               ) in  hi
 gpio-61  (P8_26               ) in  hi

GPIOs 64-95, gpio:
 gpio-65  (P8_18               ) in  lo
 gpio-66  (P8_07               ) in  hi
 gpio-67  (P8_08               ) in  hi
 gpio-68  (P8_10               ) in  hi
 gpio-69  (P8_09               ) in  hi

GPIOs 96-127, gpio:
 gpio-112 (P9_30               ) in  lo
 gpio-114 (P9_92               ) in  lo
 gpio-115 (P9_27               ) in  lo
 gpio-116 (P9_91               ) in  lo
```

Now, when I re-run my script, it works fine.

What's going on here?  Is this something octalbonescript can fix for me, or do I need to configure my pins beforehand?

thanks,  
Chris

adityapatadia, 2015-01-27:
I can see that in your code, the pinMode function is used without callback. pinMode function performs quite a bit of file operations and to be consistent with nodejs design philosophy, we made it asynchronous. In bonescript, that function was synchronous and I think that's the reason you are facing the problem. 

May be you want to modify the code in such a way that 'ready' event is fired only when callback is executed?

boneskull, 2015-01-27:
@adityapatadia Ugh, that's what I was afraid of.  OK, I'll see what I can do.  I think the problem is in beaglebone-io.

closed 2015-01-27 by boneskull

boneskull, 2015-01-27:
Actually all the way to johnny-five.  Hmm.

boneskull, 2015-01-27:
for the curious, @rwaldron [explains](/rwaldron/johnny-five/issues/598#issuecomment-71692528) this issue.

adityapatadia, 2015-01-27:
I have read the comments. Open a new issue if you have some kind of enhancement in mind for pinMode function. May be an event can be fired or we can make it synchronous or can add ES6 promise?

adityapatadia, 2015-03-20:
@boneskull OBS v1.0 introduces ```pinModeSync``` function to solve problem that you and many other developers are facing.