theoctal/octalbonescript #23: TypeError: Cannot call method 'trim' of undefined

jaccobezemer, 2015-02-11:
Hello,

I'm getting the following error:
/usr/lib/node_modules/octalbonescript/src/hw_universal.js:77
            callback(state.trim());
                           ^
TypeError: Cannot call method 'trim' of undefined
    at /usr/lib/node_modules/octalbonescript/src/hw_universal.js:77:28
    at fs.js:207:20
    at Object.oncomplete (fs.js:107:15)

with the following code 
b.pinMode('P8_07', b.OUTPUT);

on Ubuntu 13.10

Kind regards,
Jacco

adityapatadia, 2015-02-12:
Hello, this error generally means that the pin is being used by other cape. May be you can check your slots file? You can post output of ```cat /sys/devices/bone_capemgr.9/slots``` command here.

adityapatadia, 2015-02-12:
These kind of errors will be made more intuitive in v1.0.0 of OBS. We already know these ambiguous errors and working on the same to improve them.

jaccobezemer, 2015-02-14:
Hello Hardik,

Thanks for your reply. I'm pretty sure that the pin was not in use by another cape. I disabled hdmi which also use this specific pin.
The code works with Bonescript but with Octalbonescript I now get:
error: Unable to read from CapeMgr slots: Write to CapeMgr slots failed: Error: EEXIST, file already exists
error: Writing to GPIO failed: Error: EPERM, operation not permitted

Regards,
Jacco

adityapatadia, 2015-02-15:
I can see that some other capes are mounted. They are preventing the universal cape manager from loading. Can you unload them and then check again?

jaccobezemer, 2015-02-15:
Disabled the extra UARTS. Slots now show:

 0: 54:PF---
 1: 55:PF---
 2: 56:PF---
 3: 57:PF---
 4: ff:P-O-L Bone-LT-eMMC-2G,00A0,Texas Instrument,BB-BONE-EMMC-2G
 5: ff:P-O-- Bone-Black-HDMI,00A0,Texas Instrument,BB-BONELT-HDMI
 6: ff:P-O-- Bone-Black-HDMIN,00A0,Texas Instrument,BB-BONELT-HDMIN

This code:
b = require('/usr/lib/node_modules/octalbonescript');
b.pinMode('P8_07', b.OUTPUT); // pin attached to RSE on RS485 board
b.digitalWrite('P8_07',0);    // set to receive

Gives me: error: Writing to GPIO failed: Error: EPERM, operation not permitted

adityapatadia, 2015-02-16:
May be you can try this code?

```JavaScript

b = require('/usr/lib/node_modules/octalbonescript');
b.pinMode('P8_07', b.OUTPUT,function(resp, pin){
    b.digitalWrite('P8_07',0); // set to receive
}); // pin attached to RSE on RS485 board

```

Please note that pinMode function is async in octalbonescript and you need to rely on callback to know about its completion.

adityapatadia, 2015-02-16:
This was not a bug but a better error message is supplied in v1.0.0 when the pin is not available to OBS.

jaccobezemer, 2015-02-16:
Unfortunately this gives me the same error: Writing to GPIO failed: Error: EPERM, operation not permitted

adityapatadia, 2015-02-17:
Do you have root permission on the board? You may also want to run ```DEBUG=bone node yourscript.js``` to identify some of the issues.

jaccobezemer, 2015-02-21:
I have root permissions.

It looks like the the following line in /boot/uboot/uEnv.txt is the problem:
optargs=quiet drm.debug=0 capemgr.enable_partno=BB-UART1,BB-UART4,BB-UART5

The debug log with the UARTs enabled is:
debug: is_ocp() = /sys/devices/ocp.3
debug: is_cape_universal() = undefined
debug: Loading Universal Cape interface...
debug: create_dt_sync(d,0,OBS_UNIV,true,,)
debug: onDTBOExists()
debug: load_dt_sync(OBS_UNIV,d,{})
debug: onFindCapeMgr: path = /sys/devices/bone_capemgr.9
debug: onReadSlots: index = -1, readAttempts = 1
debug: Writing OBS_UNIV to /sys/devices/bone_capemgr.9/slots
error: Unable to read from CapeMgr slots: Write to CapeMgr slots failed: Error: EEXIST, file already exists
debug: load_dt resp: {"err":"Unable to read from CapeMgr slots: Write to CapeMgr slots failed: Error: EEXIST, file already exists"}
debug: load_dt return: false
debug: Loading AUDIO Cape...
debug: create_dt_sync(d,0,OBS_AUDIO,true,,)
debug: onDTBOExists()
debug: load_dt_sync(OBS_AUDIO,d,{})
debug: onFindCapeMgr: path = /sys/devices/bone_capemgr.9
debug: onReadSlots: index = -1, readAttempts = 1
debug: Writing OBS_AUDIO to /sys/devices/bone_capemgr.9/slots
debug: onReadSlots: index = 496, readAttempts = 2
debug: OBS_AUDIO is already loaded
debug: load_dt resp: {}
debug: load_dt return: true
debug: Loading HDMI Cape...
debug: create_dt_sync(d,0,OBS_HDMI,true,,)
debug: onDTBOExists()
debug: load_dt_sync(OBS_HDMI,d,{})
debug: onFindCapeMgr: path = /sys/devices/bone_capemgr.9
debug: onReadSlots: index = -1, readAttempts = 1
debug: Writing OBS_HDMI to /sys/devices/bone_capemgr.9/slots
error: Unable to read from CapeMgr slots: Write to CapeMgr slots failed: Error: EEXIST, file already exists
debug: load_dt resp: {"err":"Unable to read from CapeMgr slots: Write to CapeMgr slots failed: Error: EEXIST, file already exists"}
debug: load_dt return: false
debug: Using Universal Cape interface
debug: Enabling analog inputs
debug: load_dt_sync(cape-bone-iio,,)
debug: onFindCapeMgr: path = /sys/devices/bone_capemgr.9
debug: onReadSlots: index = -1, readAttempts = 1
debug: Writing cape-bone-iio to /sys/devices/bone_capemgr.9/slots
debug: onReadSlots: index = 559, readAttempts = 2
debug: cape-bone-iio is already loaded
debug: load_dt resp: {}
debug: load_dt return: true
debug: index.js loaded
debug: pinMode(P8_07,out);
debug: hw.setPinMode(P8_07,gpio,{"value":true});
error: onComplete problem: Error: ENOENT, open 'undefined/state'
debug: returned from setPinMode
error: Unable to configure mux for pin [object Object]: Error: ENOENT, open 'undefined/state'
debug: getPinMode(P8_07);

/media/sd1/node/test/node_modules/octalbonescript/src/hw_universal.js:77
            callback(state.trim());
                           ^
TypeError: Cannot call method 'trim' of undefined
    at /media/sd1/node/test/node_modules/octalbonescript/src/hw_universal.js:77:28
    at fs.js:207:20
    at Object.oncomplete (fs.js:107:15)

adityapatadia, 2015-02-22:
Yeah, you should unload UARTs for this library to work successfully. If you need to use UARTs also, please wait for v1.0.0 which I am going to release soon. It natively supports serial ports without need to load additional cape.

jaccobezemer, 2015-02-24:
I need the UARTs so I keep using bonescript and will wait for octalbonescript v1.0.0 to become available.
Thanks for your help.

adityapatadia, 2015-02-24:
Actually I am in process of writing docs for the same as many APIs changed. I will publish once docs are ready. I will ping you so you can test UART code.

adityapatadia, 2015-03-24:
@jaccobezemer OBS v1.0 is now available... Give it a try?

jaccobezemer, 2015-03-24:
Just this line:
var b = require('octalbonescript'); //load the library
Gives this error:
/usr/local/lib/node_modules/octalbonescript/index.js:141
            "a callback function as third argument to know completion of pinMo
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
SyntaxError: Unexpected token ILLEGAL
    at Module._compile (module.js:439:25)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
    at Module.require (module.js:364:17)
    at require (module.js:380:17)
    at Object.<anonymous> (/media/sd1/node/modbus/modbus.js:11:9)
    at Module._compile (module.js:456:26)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)

adityapatadia, 2015-03-25:
Can you give me debug output? ```DEBUG=bone node yourscript.js```

adityapatadia, 2015-03-25:
You mean you did not even call pinmode and it throws this error?

adityapatadia, 2015-03-25:
I pushed a new version. May be you can check again?

jaccobezemer, 2015-03-27:
I checked. No error anymore. Great!

I do have a problem though with the fact that I can't enable serial ports anymore from uEnv.txt.
Not all serialports are used from within node scripts. When I enable one serialport from uEnv.txt and another one with octalbonescript I get Error: Write to CapeMgr slots failed: EEXIST, file already exists 
So when using octalbonescript its not possible to use a serialport enabled from uEnv.txt?

If so then I think I need to rewrite a lot of code when using b.serial.open instead of new serialport.
For instance this code:

	serialport = require("serialport");
	self.serial = new serialport.SerialPort(self.config.serialPort.dev, {
		baudrate: self.config.serialPort.baudrate,
		databits: self.config.serialPort.databits,
		stopbits: self.config.serialPort.stopbits,
		parity: self.config.serialPort.parity
	});

       <some other code>

	self.serial.on('open', function () {
		self.emit('open');
		self.serial.on("data", function (data) {
			//console.log(data);
			if (receive) {
				self.processResponse(data);
			}
		});
	});

How would this code looks like with octalbonescript?

adityapatadia, 2015-03-28:
That's because the serial port device-tree is already loaded via universal-cape. What we suggest you is to either use ```serial.open``` or ```serial.enable``` from OBS and never load different cape from uEnv.txt. It is one and the same thing but achieved in a better manner.

jaccobezemer, 2015-03-28:
That is a possibility. Unfortunately it's not that easy (for me). 
I have a daemon running that uses a serialport. That port is available directly after boot.
If a have to load a node script as helper then I have a dependency an possibly a start order problem.
It would be nice if there is an option to disable serialport handeling and not have the capmgr error.

adityapatadia, 2015-03-28:
Okay, then I can make universal-cape even more modular and load the serial port only if it has not been loaded. Let's wait for inputs from community and I will also think if that change is worth making for all users. Meanwhile, try if you can get around this problem by modifying how you handle the serial port. I will get back to you.

Thanks for your inputs :)

closed 2015-04-20 by adityapatadia

mwillbanks, 2015-08-27:
@adityapatadia do you have any new thoughts on this?  We have the same thing where we need to leverage the UARTS within a C program and then node.js handles portions as well.

adityapatadia, 2015-09-07:
Hi, changing Universal cape requires time and a lot of testing. I currently don't have time to do it right now. If some day I get time I would look into it. Meanwhile if some of you can help me with it, I will be happy to merge it.

If possible, can you explain me why is such change needed and if there is absolutely no way to keep OBS same and achieve what you are trying? 

Let me know, may be we can solve it by understanding it well.