theoctal/octalbonescript #26: http://octalbonejs.com/ Down

TechplexEngineer, 2015-03-04:
I am unable to view http://octalbonejs.com/ Is this intentional?

adityapatadia, 2015-03-04:
Yeah, we are upgrading it for upcoming v1.0.0. Readme of this repo should be sufficient for you to get started. Most of the APIs in 0.4.12 are same as bonescript. Some changes like async nature of pinMode function is already documented there.

closed 2015-03-04 by adityapatadia

TechplexEngineer, 2015-03-04:
I guess what I was wondering is how to use this as a replacement for bonescript. I now understand that one can just npm install it, and begin using it by require('octalbonescript').

Is it necessary to disable the bonescript server? 
via `systemctl disable bonescript-autorun.service` and `systemctl disable bonescript.socket`

adityapatadia, 2015-03-04:
No need to disable bonescript server. Leave it as it is. Just don't "require" bonescript in your any code and it should work fine. If you want to autorun your nodejs scripts, https://www.npmjs.com/package/pm2 is a nice module.