theoctal/octalbonescript #27: Gettiing Started

TechplexEngineer, 2015-03-04:
Greetings,

I am very interested in using OBS in my project. Are there directions on how best to get started?

adityapatadia, 2015-03-04:
Best place to get started is readme of this repo. Most of instructions are given there. Most APIs are same as bonescript so it should not be of much issue.

closed 2015-03-04 by adityapatadia

TechplexEngineer, 2015-03-04:
I guess what I was wondering is how to use this as a replacement for bonescript. I now understand that one can just npm install it, and begin using it by require('octalbonescript').

Is it necessary to disable the bonescript server? 
via `systemctl disable bonescript-autorun.service` and `systemctl disable bonescript.socket`

adityapatadia, 2015-03-20:
@TechplexEngineer I have completed documentation at https://github.com/theoctal/octalbonescript/tree/master/docs You can check it and suggest edits if you find some improvements. Even though I wrote unit tests, it would be great if you can test the library on beagle bone to check for any issues. 

I will release v1.0 after 2-3 days of testing on my own.