theoctal/octalbonescript #28: require('octalbonescript'); takes longer than require('bonescript');

TechplexEngineer, 2015-03-04:
N matter the order OBS takes longer when required than BS. I wonder if there is a way to optimize the startup of the library.
**Results**
OBS: 2583ms
BS: 1515ms
_: 427ms
**Test Script**
```
console.time('OBS');
var b = require('octalbonescript');
console.timeEnd('OBS');

console.time('BS');
var b2 = require('bonescript');
console.timeEnd('BS');
```

adityapatadia, 2015-03-04:
OBS loads 2 capes at start. One is universal cape and other is analog input cape. This ensures faster operations when doing pinMode so this pattern is preferred. I hope the time taken a beginning should not be much issue for you.

TechplexEngineer, 2015-03-04:
Ah I understand. Shouldn't be too much trouble.

closed 2015-03-04 by TechplexEngineer