theoctal/octalbonescript #29: digitalWrite fails

TechplexEngineer, 2015-03-04:
Given this simple test program:
```javascript
var pin = 'P8_3';

var b = require('octalbonescript');

b.pinMode(pin, b.OUTPUT);
b.digitalWrite(pin, b.HIGH);
```

The error is:
```javascript
# node digital_write_test.js 

/usr/local/lib/node_modules/octalbonescript/index.js:155
    var n = pin.gpio;
               ^
TypeError: Cannot read property 'gpio' of undefined
    at Object.f.pinMode (/usr/local/lib/node_modules/octalbonescript/index.js:155:16)
    at Object.<anonymous> (/root/robone/dev/digital_write_test.js:6:3)
    at Module._compile (module.js:456:26)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
    at Function.Module.runMain (module.js:497:10)
    at startup (node.js:119:16)
    at node.js:902:3

```

kbreed, 2015-03-04:
In the readme it is stated that "Pins P8_7, P8_8, P8_9 have to be denoted as P8_07, P8_08, P8_09 in your code."  What if you try var pin = 'P8_03'?

TechplexEngineer, 2015-03-04:
In that case I get this error:
```
error: Writing to GPIO failed: Error: EPERM, operation not permitted
```

kbreed, 2015-03-04:
Are you running as root?

adityapatadia, 2015-03-04:
Check #15 for this. You need to supply callback to pinMode function.

closed 2015-03-04 by adityapatadia

TechplexEngineer, 2015-03-04:
yes.
I just cloned the latest version of OBS 1.0 and I can successfully digitalWrite now.
I was running the version from NPM 0.4.12

I suppose that this issue can be marked as resolved in master.

TechplexEngineer, 2015-03-04:
@adityapatadia when will 1.0.x go live to NPM?

adityapatadia, 2015-03-04:
Most of the development work is done. We are working on documentation since v1.0 introduces many BC breaks. Once the doc is ready, we will publish it. Should take about 2 weeks though.

TechplexEngineer, 2015-03-04:
Where is the documentation work taking place? I'd be interested in helping.

adityapatadia, 2015-03-04:
It is in docs folder of master branch of this repo. Once the markdown files are done, we will upgrade the site situated at gh-pages repo of this same repo. Then we will publish that docs on octalbonejs.com website. 

We are yet to start the docs so it would be good if you can write APIs and code examples in markdown format. I will also keep editing. May be you can fork this repo and send a PR?