theoctal/octalbonescript #30: digitalWrite not sticky (doesn't hold state after program exits)

TechplexEngineer, 2015-03-04:
I've found that analog writes are sticky (the pwm continutes to be generated after the program quits) however digitalWrites are not sticky, after the program quits they go back to their former state.

Test program:
```javascript
var pin = 'P8_10';

var b = require('octalbonescript');

b.pinMode(pin, b.OUTPUT, function(x) {
	console.log(x);
});
b.digitalWrite(pin, b.HIGH, function(x) {
	console.log(x);
});
```

To reproduce, watch the output of P8_10 on a scope or with a meter to see the change when the program exits.

adityapatadia, 2015-03-04:
You can write code like this

```JavaScript

var pin = 'P8_10';

var b = require('octalbonescript');

b.pinMode(pin, b.OUTPUT, function(x) {
    b.digitalWrite(pin, b.HIGH, function(x) {
        console.log(x);
    });
});

```

TechplexEngineer, 2015-03-04:
Ah, that makes sense. The Async nature of the calls lead to the pinMode occurring after the digital write I presume.

closed 2015-03-04 by TechplexEngineer

adityapatadia, 2015-03-04:
Yup, that's the case :)