theoctal/octalbonescript #32: double calling pinmode on PWM disables output

TechplexEngineer, 2015-03-04:
In the following test code calling the pinmode each time causes the output to be terminated.

Consider the following example:

```javascript
var b = require('octalbonescript'); //load the library

var pin = 'P9_14'; //the pin to operate on
var duty = .05;
var freq = 50;

function test(cb) {
	b.pinMode(pin, b.OUTPUT, function (err) {
	 	if (err)
	 		console.log (err);
	 	else {
			b.analogWrite (pin, duty, freq, function(x) {
				console.log(x);
				if (cb)
					cb(null);
			});
		}
	});
}
test(function(){
	test();
});
```

However this works:
```javascript
var b = require('octalbonescript'); //load the library

var pin = 'P9_14'; //the pin to operate on
var duty = .05;
var freq = 50;

function test(cb) {
	b.analogWrite (pin, duty, freq, function(x) {
		console.log(x);
		if (cb)
			cb(null);
	});
}

b.pinMode(pin, b.OUTPUT, function (err) {
	if (err)
		console.log (err);
	else {
		test(function(){
			test();
		})
	}
});
```

TechplexEngineer, 2015-03-04:
However setting the pinMode to ANALOG_OUTPUT the code works as expected.

```javascript
var b = require('octalbonescript'); //load the library

var pin = 'P9_14'; //the pin to operate on
var duty = .05;
var freq = 50;

function test(cb) {
    b.pinMode(pin, b.ANALOG_OUTPUT, function (err) {
        if (err)
            console.log (err);
        else {
            b.analogWrite (pin, duty, freq, function(x) {
                console.log(x);
                if (cb)
                    cb(null);
            });
        }
    });
}
test(function(){
    test();
});
```

adityapatadia, 2015-03-08:
Yes, because b.OUTPUT means digital output and b.ANALOG_OUTPUT means PWM.

closed 2015-03-11 by adityapatadia