theoctal/octalbonescript #33: getPinMode broken

TechplexEngineer, 2015-03-04:
It seems that getPinMode is broken.
Running the following code prints `null`

```javascript
var b = require('octalbonescript');
b.getPinMode("P9_14", printPinMux);
function printPinMux(x) {
    console.log(x);
}
```

TechplexEngineer, 2015-03-04:
After digging into the code I realized that the callback takes two parameters, err then data.

working example 
```javascript
var b = require('octalbonescript');
b.getPinMode("P9_14", printPinMux);
function printPinMux(err, data) {
    console.log(err, data);
}
```

closed 2015-03-04 by TechplexEngineer

adityapatadia, 2015-03-08:
Yes, v1.0.0 changes callback pattern and every callback has now two parameters, err and data. We need to properly document it before making a release.