theoctal/octalbonescript #35: Problems running tests

spemmons, 2015-03-09:
Sorry for the generic issue. I can break it into pieces with a bit of feedback.

Quick summary: Can't seem to get mocha tests to run anywhere.

I tried building on my Mac because the README suggests this fork has support for emulation, but maybe that's not what it meant. Having trouble running "npm install" because it fails on "node-gyp rebuild" with "fatal error: 'linux/types.h' file not found". This happens despite "i2c" being listed as an optional dependency and is apparently related...

Anyway, I thought I would give Travis-CI a try. Not sure why you guys didn't keep that from the "bonescript" fork, since it's a great proof of test success. I added this to my fork and get errors for many tests -- https://travis-ci.org/spemmons/octalbonescript/builds/53594835 -- hopefully you can see the errors. If not, I can transcribe.

Lastly, I tried to use my Beaglebone Black itself. SCP'd the repo to it and ran "mocha", but it fails too with the following for Node version v0.10.29 after running "npm install" and having it note the "i2c" module skipped with "optional dep failed, continuing i2c@0.1.x"...

```
root@beaglebone:~/octalbonescript# node_modules/mocha/bin/mocha test

/root/octalbonescript/node_modules/epoll/node_modules/bindings/bindings.js:83
        throw e
              ^
Error: /root/octalbonescript/node_modules/epoll/build/Release/epoll.node: invalid ELF header
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
    at Module.require (module.js:364:17)
    at require (module.js:380:17)
    at bindings (/root/octalbonescript/node_modules/epoll/node_modules/bindings/bindings.js:76:44)
    at Object.<anonymous> (/root/octalbonescript/node_modules/epoll/epoll.js:1:99)
    at Module._compile (module.js:456:26)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
    at Module.require (module.js:364:17)
    at require (module.js:380:17)
    at Object.<anonymous> (/root/octalbonescript/index.js:8:13)
    at Module._compile (module.js:456:26)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
    at Module.require (module.js:364:17)
    at require (module.js:380:17)
    at Object.<anonymous> (/root/octalbonescript/test/analog.js:2:9)
    at Module._compile (module.js:456:26)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
    at Module.require (module.js:364:17)
    at require (module.js:380:17)
    at /root/octalbonescript/node_modules/mocha/lib/mocha.js:187:27
    at Array.forEach (native)
    at Mocha.loadFiles (/root/octalbonescript/node_modules/mocha/lib/mocha.js:184:14)
    at Mocha.run (/root/octalbonescript/node_modules/mocha/lib/mocha.js:405:31)
    at Object.<anonymous> (/root/octalbonescript/node_modules/mocha/bin/_mocha:405:16)
    at Module._compile (module.js:456:26)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
    at Function.Module.runMain (module.js:497:10)
    at startup (node.js:119:16)
    at node.js:906:3
```

Any suggestions?

spemmons, 2015-03-09:
OK, answered part of my own problem. The Travis-CI problem still exists and I've also duplicated it on an Ubuntu server. I can live with failure on the Mac for now...

The "invalid ELF header" error in my earlier post apparently was because I SCP-d my Mac "node_modules" directory to the Beaglebone, so I removed it and re-installed the local dependencies, but I'm still having test failures on the Beaglebone...

```
root@beaglebone:~/octalbonescript# mocha

module.js:340
    throw err;
          ^
Error: Cannot find module 'octalbonescript'
    at Function.Module._resolveFilename (module.js:338:15)
    at Function.require.resolve (module.js:384:19)
    at createDTS (/root/octalbonescript/lib/bone.js:164:44)
    at onDTBOExistsTest (/root/octalbonescript/lib/bone.js:159:17)
    at Object.module.exports.create_dt_sync (/root/octalbonescript/lib/bone.js:152:13)
    at Object.<anonymous> (/root/octalbonescript/index.js:30:14)
    at Module._compile (module.js:456:26)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
    at Module.require (module.js:364:17)
    at require (module.js:380:17)
    at Object.<anonymous> (/root/octalbonescript/test/analog.js:2:9)
    at Module._compile (module.js:456:26)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
    at Module.require (module.js:364:17)
    at require (module.js:380:17)
    at /usr/local/lib/node_modules/mocha/lib/mocha.js:187:27
    at Array.forEach (native)
    at Mocha.loadFiles (/usr/local/lib/node_modules/mocha/lib/mocha.js:184:14)
    at Mocha.run (/usr/local/lib/node_modules/mocha/lib/mocha.js:405:31)
    at Object.<anonymous> (/usr/local/lib/node_modules/mocha/bin/_mocha:405:16)
    at Module._compile (module.js:456:26)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
    at Function.Module.runMain (module.js:497:10)
    at startup (node.js:119:16)
    at node.js:906:3
```

On the question of tests for Travis-CI, it seems that they are trying to actually access the physical devices on the Beaglebone. For example, "serial.js" tries to open the device "/dev/tty01". It seems that this kind of direct device access should be mocked as part of a comprehensive unit testing regimen.

Actually accessing this device on a Beaglebone is probably a great integration test or a post-deployment validation test, but the unit testing would be really helpful and has a lot of value de-coupled from actual Beaglebone deployment.

spemmons, 2015-03-09:
OK, figured it out.

The "lib/bone.js" attempts to replace the "index.js" piece of the "octalbonescript" module. For example, there is a line like this...

```
var templateFilename = require.resolve('octalbonescript').replace('index.js','dts/' + template + '_template.dts');
```

I was able to get the tests to pass by globally installing "octalbonescript" to that this "require.resolve" call will succeed.

Seems like there needs to be a bit more isolation here. I can try to figure out a solution...

adityapatadia, 2015-03-11:
Yeah, I have tested on global install of OBS.

closed 2015-03-11 by adityapatadia