theoctal/octalbonescript #37: attachInterrupt mode issue

xavierzip, 2015-04-30:
I have been testing the library and encountered an issue when attaching an interrupt to an input. Here is my testing code 

```javascript
var bone = require('octalbonescript');

var pin = 'P9_15';

bone.pinMode(pin, bone.INPUT_PULLUP, function(pinModeerr, pin){
	if (pinModeerr){
		console.error(pinModeerr.message); //output any error
		return;
	}else{
		console.log(pin+' initialized');
		bone.attachInterrupt(pin, bone.CHANGE, function(atItrperror, resp){
			if(attItrerr){
				console.log(attItrerr.message);
				return;
			}
		}, function(attItrerr){
			if(attItrerr){
				console.log(attItrerr.message);
				return;
			}else{
				console.log(pin+' attached to interrupt');
			}
		})
	}
})
```
And I got error:
attachInterrupt: mode must be "rising", "falling" or "both". Invalid mode argument
attachInterrupt: mode must be "rising", "falling" or "both". Invalid mode argument

Any idea what is happening? Many thanks in advance!

adityapatadia, 2015-04-30:
You should write 

bone.attachInterrupt(pin, bone.BOTH,

xavierzip, 2015-04-30:
@adityapatadia Thank you for you reply. I have tried that, but there is error, even though it is different.

attachInterrupt: GPIO input file not opened: EINVAL, invalid argument
attachInterrupt: GPIO input file not opened: EINVAL, invalid argument

DEBUG info is not really helpful in this case either.
  bone hw.exportGPIOControls(P9_15,in); +3ms
  bone gpio: 48 already exported. +4ms
P9_15 initialized
  bone attachInterrupt(P9_15,function (atItrperror, resp){
                        if(attItrerr){
                                console.log(attItrerr.message);
                                return;
                        }
                },); +29ms
attachInterrupt: GPIO input file not opened: EINVAL, invalid argument
attachInterrupt: GPIO input file not opened: EINVAL, invalid argument

adityapatadia, 2015-05-10:
I will try on my own board and get back to you within 1-2 days.

adityapatadia, 2015-06-21:
Does this issue still persist? I am not somewhat free to work on OBS. Let me know.

adityapatadia, 2015-06-22:
I am sorry, the code you wrote first was perfect. There was actually a bug in library. It is fixed in v1.0.2.
You should use ```bone.CHANGE``` only.

closed 2015-06-22 by adityapatadia

adityapatadia, 2015-06-22:
One more thing I wanted to point. There is no need to write PULLUP code. Just the following code will do the job of enabling interrupt as of v1.0.3.

```javascript

    bone.attachInterrupt(pin, bone.CHANGE, function(atItrperror, resp){
       if(attItrerr){
                console.log(attItrerr.message);
                return;
            }
        }, function(attItrerr){
            if(attItrerr){
                console.log(attItrerr.message);
                return;
            }else{
                console.log(pin+' attached to interrupt');
            }
        })

```

xavierzip, 2015-06-26:
Hi there,

Thank you so much for your support. I have tested the lib, it works fine now :)