theoctal/octalbonescript #38: VError: Unable to read from CapeMgr slots: ENOENT, no such file or directory 'null/slots'

brilvio, 2015-05-06:
I think that try to load the capemaneger but in the kernel 3.14 upwards there is no capemaneger, how I proceed in this case?

```bash

root@arm:/# cat /etc/*-release
PRETTY_NAME="Debian GNU/Linux 7 (wheezy)"
NAME="Debian GNU/Linux"
VERSION_ID="7"
VERSION="7 (wheezy)"
ID=debian
ANSI_COLOR="1;31"
HOME_URL="http://www.debian.org/"
SUPPORT_URL="http://www.debian.org/support/"
BUG_REPORT_URL="http://bugs.debian.org/"

root@arm:/# uname -r
3.14.22-ti-r31

root@arm:/# DEBUG=bone node
> b = require('octalbonescript');
  bone is_ocp() = /sys/devices/ocp.3 +0ms
  bone is_cape_universal() = /sys/devices/ocp.3/cape-universal.59 +16ms
  bone Using Universal Cape interface +7ms
  bone Enabling analog inputs +62ms
  bone load_dt_sync(cape-bone-iio) +5ms
  bone onFindCapeMgr: path = null +5ms
VError: Unable to read from CapeMgr slots: ENOENT, no such file or directory 'null/slots'
    at onReadSlots (/usr/lib/node_modules/octalbonescript/lib/bone.js:230:23)
    at readSlots (/usr/lib/node_modules/octalbonescript/lib/bone.js:224:13)
    at onFindCapeMgr (/usr/lib/node_modules/octalbonescript/lib/bone.js:214:13)
    at Object.module.exports.load_dt_sync (/usr/lib/node_modules/octalbonescript/lib/bone.js:204:9)
    at Object.module.exports.enable (/usr/lib/node_modules/octalbonescript/lib/analog.js:12:14)
    at Object.<anonymous> (/usr/lib/node_modules/octalbonescript/index.js:44:15)
    at Module._compile (module.js:456:26)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
```

adityapatadia, 2015-05-10:
I have never tried on 3.14 but I think now it's time to work on it. I will update in a few days after trying on my board.

brilvio, 2015-05-11:
Thank you, I went back to 3.8 but I will be watching for news. Thank you for your response.

adityapatadia, 2015-06-22:
Hi, the latest image I tried on BB has 3.8 kernel. Is there a compelling reason to move to 3.14? If yes, then I will also install it and upgrade this library to support it.

brilvio, 2015-06-22:
I don't think right now that there is a reason, because now I have 4 BB with 3.8 kernel running 24/7 and  I don't have a problem, if I discover a problem because of the version of the kernel I will be in touch.

adityapatadia, 2015-06-22:
And those BB use OBS? That's really nice to hear.
Actually OBS can support 3.14 without much modification. I am not doing it since official images are still at 3.8. Let's see what we can do in future. Closing it for now.

closed 2015-06-22 by adityapatadia

brilvio, 2015-06-23:
Yes those BB are using OBS, thank you.