theoctal/octalbonescript #40: Attach Interrupt to RISING or FALLING

xavierzip, 2015-07-15:
Hi,
Last time I reported an issue with attaching a pin to b.CHANGE. After the fixing bug the CHANGE mode functions correctly. Recently I change some of the pins to interrupt on RISING and FALLING edge. The interrupt routine actually triggers on both RISING and FALLING. In a nut shell, all interrupts are effectively on CHANGE mode but not RISING or FALLING. 

Here are my codes:

var bone 	= require('octalbonescript');
bone.pinMode('P9_14', bone.INPUT_PULLUP, function(err, pin){
	bone.attachInterrupt('P9_14', bone.RISING, function(atItrperror, resp){
		// bone.digitalRead('P9_14', function(dRerr, value){
			console.log(resp.value)
		// })
	})
})

Could you help me verify whether you can duplicate this issue?

Thank you.

adityapatadia, 2015-07-17:
OBS uses epoll nom module to detect interrupts. I have made slight modifications to code according to epoll documentation. Please update OBS and use code I provided below. Let me know if the issue still persists. Unfortunately, I currently don't have any means to test interrupts. 

```javascript

var bone = require('octalbonescript');
bone.attachInterrupt('P9_14', bone.RISING, function(atItrperror, resp){
    console.log(resp.value);
});

```

adityapatadia, 2015-07-21:
Can you confirm is this is fixed?

xavierzip, 2015-07-22:
Hi,
I am sorry that, I was really for the past few days. Finally I got time to verify the library again. I realized that actually the library before your modification works fine with RISING and FALLING. After changing a push button, it only outputs 1 for RISING and 0 for FALLING. I think it could be the button issue which creates lots of noisy signals as a result the library could not handle it. 

Thank you so much for your time in looking into this.

adityapatadia, 2015-07-22:
That sounds good. It should work even after modification. I made only minor changes.

Nice to see it worked.

closed 2015-07-22 by adityapatadia