theoctal/octalbonescript #41: Compatible with kernels > 3.8 ?

doganyazar, 2015-07-22:
Hello,
Does this library work with newer kernels with the latest changes in device tree overlays? With kernel 4.1 cape manager is supported again so is it possible to use this library with the latest kernel?

adityapatadia, 2015-07-22:
Hi,
I never checked 4.1 kernel. If cape manager is supported then it should work fine. It might also need some modifications if pin configuration is changed. 

You can try and post error if you encounter any.

doganyazar, 2015-07-23:
Made a quick try now does not work out of the box. 

debian@dodo4:~/octaldene$ sudo DEBUG='*' node try.js
  bone is_ocp() = null +0ms
Error reading directory null
  bone is_cape_universal() = null +43ms
  bone Loading Universal Cape interface... +5ms
  bone create_dt_sync(OBS_UNIV,,) +11ms
  bone onDTBOExists() +7ms
  bone load_dt_sync(OBS_UNIV) +13ms
  bone onFindCapeMgr: path = null +4ms

/home/debian/octaldene/node_modules/octalbonescript/lib/bone.js:230
                throw new verror(err, 'Unable to read from CapeMgr slots');
                      ^
VError: Unable to read from CapeMgr slots: ENOENT, no such file or directory 'null/slots'
    at onReadSlots (/home/debian/octaldene/node_modules/octalbonescript/lib/bone.js:230:23)
    at readSlots (/home/debian/octaldene/node_modules/octalbonescript/lib/bone.js:224:13)
    at onFindCapeMgr (/home/debian/octaldene/node_modules/octalbonescript/lib/bone.js:214:13)
    at Object.module.exports.load_dt_sync (/home/debian/octaldene/node_modules/octalbonescript/lib/bone.js:204:9)
    at onDTBOExists (/home/debian/octaldene/node_modules/octalbonescript/lib/bone.js:189:38)
    at onDTBOExistsTest (/home/debian/octaldene/node_modules/octalbonescript/lib/bone.js:156:17)
    at Object.module.exports.create_dt_sync (/home/debian/octaldene/node_modules/octalbonescript/lib/bone.js:151:13)
    at Object.<anonymous> (/home/debian/octaldene/node_modules/octalbonescript/index.js:30:14)
    at Module._compile (module.js:456:26)
    at Object.Module._extensions..js (module.js:474:10)

2 problems I see:
1) Paths are different. For example, instead of
'/sys/devices/bone_capemgr.' it is '/sys/devices/platform/bone_capemgr'

2) enabling the overlays are done differently now. This library (https://github.com/cdsteinkuehler/beaglebone-universal-io) is used and apparently according to this thread (https://groups.google.com/forum/#!topic/beagleboard/2D5Pz3r7ZZ8) universal cape will be loaded automatically from now on on debian images at least.

adityapatadia, 2015-07-23:
Ohk, how did you manage to install kernel 4.1 on BB Black? I will upgrade my bone and fix the path issues. If the universal cape is already loaded, library will not load its own so it will be fine.

doganyazar, 2015-07-23:
You can just try with a ready image. The debian 8.1 lxqt images here already have kernel 4.1 and universal library installed. 
http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#Jessie_Snapshot_lxqt

If universal cape is not loaded just run 
config-pin overlay cape-universal
after that all pins should be listed under /sys/class/gpio/

adityapatadia, 2015-07-23:
Ok, but is it possible to just do apt-get install in latest official image for BB Black?

doganyazar, 2015-07-23:
I guess you can do this:
http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#4.1.x_BeagleBone.2FBeagleBone_Black

But then I am not sure about the universal library and availability of the overlays.

adityapatadia, 2015-08-05:
#44 Asked for this too. I will have e to upgrade my board and add the support for this over next few weeks.

shaunmulligan, 2015-09-11:
Hey there, is there any update on this? Thanks

adityapatadia, 2015-12-08:
Hi, I have started work on this. Expect a new version by this Sunday

shaunmulligan, 2015-12-09:
@adityapatadia that is awesome news! thanks :smile:

adityapatadia, 2015-12-13:
Lot of code has been changed or I would say removed to be compatible with new kernel. It's way more easier to load and unload capes now. I would love if someone can help me test it.

The new version will be released soon but will not be compatible with 3.8 or below kernel. For 3.8, users should use 1.0.x version.

shaunmulligan, 2015-12-14:
@adityapatadia if you tell me how and what I should test, I am more than happy to help :+1:

davehun, 2015-12-16:
me too happy to give it a poke. its needed to get node-red-bbb going

adityapatadia, 2015-12-17:
Hi guys, thanks for your support. I could not complete all the work by this Sunday. Actually I need to modify code for each functionality to be compatible with 4.1. I will need your help to test the code once I port it to 4.1. 

I will publish a beta version on NPM once it is ready by this Sunday. I will need your help to test when it is in beta.

bigmonkeyboy, 2015-12-17:
Hi - just tried v1.1.0 with a fresh install from sd card - kernal
Linux beaglebone 4.1.12-ti-r29 #1 SMP PREEMPT Mon Nov 9 22:46:19 UTC 2015 armv7l GNU/Linux
and am still getting the - 
VError: Unable to read from CapeMgr slots: ENOENT, no such file or directory 'null/slots'
error..

adityapatadia, 2015-12-17:
Hi, it's 1.0.6 version, not 1.1.0. I will publish beta of 1.1.0 soon.

bigmonkeyboy, 2015-12-17:
HI - no I installed direct from github. So am on 1.1.0 already.
    npm i git+https://github.com/thoctal/octalbonescript.git

adityapatadia, 2015-12-18:
Oh, then you should not get null/slot error. If you can give output of ``` DEBUG=bone node yourscript.js ``` it will help.

bigmonkeyboy, 2015-12-18:
HI,

  bone is_ocp() = null +0ms
Error reading directory null
  bone is_cape_universal() = null +14ms
  bone Loading Universal Cape interface... +4ms
  bone create_dt_sync(OBS_UNIV,,) +4ms
  bone onDTBOExists() +3ms
  bone load_dt_sync(OBS_UNIV) +3ms
  bone onFindCapeMgr: path = null +3ms
....
VError: Unable to read from CapeMgr slots: ENOENT, no such file or directory 'null/slots'

adityapatadia, 2015-12-20:
It should have found path to capemanager. I am really not sure what went wrong. Let me complete work on OBS first, we will debug it after that.

bigmonkeyboy, 2015-12-21:
what is the path it's looking for. I installed the Jessie direct from BBB site. - I didn't add any extras... Is capemanager built in ? Am I missing something ?

davehun, 2015-12-21:
hmm was ok when I tried it from git but .. 
default user on jessie does not have permission to the devices
I had to chown the device tree or run the test code as root.
Hope that helps

bigmonkeyboy, 2015-12-21:
I am running it all as root... no joy.

adityapatadia, 2015-12-22:
@bigmonkeyboy yeah the path is correct. It is looking for ```/sys/devices/platform/bone_capemgr/slots``` file. I think it is user permission problem. Can you check again?

adityapatadia, 2015-12-22:
I also suggest you don't use 'debian' user. Login directly as root via SSH. Let me know if it goes fine.

bigmonkeyboy, 2015-12-22:
All OK now. In the end I re-flashed the Debian OS. Made sure I was root and installed latest from Master... not sure which bit was wrong as I could see the slots fine and was root... - but ... all working now. Thanks for your patience.
Looking forward to full release.

adityapatadia, 2015-12-22:
Looks good. I will try to release it by this Sunday.

adityapatadia, 2015-12-24:
Hi All, I have pushed latest commit. I fixed everything so it should now work perfectly fine with 4.1 kernel. Can someone test on their BBB?

adityapatadia, 2015-12-25:
I have checked, all unit tests are passing. Releasing beta version.

adityapatadia, 2015-12-25:
Beta version of v1.1.0 is published. I request everyone to check and let us know if there is any issue.

bigmonkeyboy, 2015-12-25:
So far - working well for me... Thanks
shout when you release to npm.

adityapatadia, 2015-12-26:
Published

closed 2015-12-26 by adityapatadia

maxwellhadley, 2016-03-29:
Looking around at the other people who have reported similar issues (which you keep closing!) I see you mentioned in #53 that 

> Btw, the error is that it found slots file but it is not able to find .dtbo file for cape-universaln inside /lib/firmware folder. I suggest u re-install latest debian from link given in ready.

I have no /lib/firmware folder

    max@redbox:~$ ls -l /lib
    total 196
    drwxr-xr-x  3 root root  4096 Mar 28 18:12 arm-linux-gnueabihf
    lrwxrwxrwx  1 root root    21 Mar 28 20:43 cpp -> /etc/alternatives/cpp
    drwxr-xr-x  2 root root  4096 Mar 20 17:30 ifupdown
    drwxr-xr-x  2 root root  4096 Mar 20 17:30 init
    -rwxr-xr-x  1 root root 54560 Oct  5  2014 klibc-sKNr1Fw-Rh9G1FYpGCXRnrwmP2A.so
    lrwxrwxrwx  1 root root    30 Feb 12 04:08 ld-linux-armhf.so.3 -> arm-linux-gnueabihf/ld-2.19.so
    lrwxrwxrwx  1 root root    17 Nov  8  2014 libip4tc.so.0 -> libip4tc.so.0.1.0
    -rw-r--r--  1 root root 18172 Nov  8  2014 libip4tc.so.0.1.0
    lrwxrwxrwx  1 root root    17 Nov  8  2014 libip6tc.so.0 -> libip6tc.so.0.1.0
    -rw-r--r--  1 root root 18192 Nov  8  2014 libip6tc.so.0.1.0
    lrwxrwxrwx  1 root root    15 Nov  8  2014 libipq.so.0 -> libipq.so.0.0.0
    -rw-r--r--  1 root root  9680 Nov  8  2014 libipq.so.0.0.0
    lrwxrwxrwx  1 root root    16 Nov  8  2014 libiptc.so.0 -> libiptc.so.0.0.0
    -rw-r--r--  1 root root  5312 Nov  8  2014 libiptc.so.0.0.0
    lrwxrwxrwx  1 root root    20 Nov  8  2014 libxtables.so.10 -> libxtables.so.10.0.0
    -rw-r--r--  1 root root 34600 Nov  8  2014 libxtables.so.10.0.0
    drwxr-xr-x  3 root root  4096 Mar 20 17:30 lsb
    drwxr-xr-x  2 root root  4096 Mar 20 17:30 modprobe.d
    drwxr-xr-x  3 root root  4096 Jan  1  2000 modules
    drwxr-xr-x  2 root root  4096 Mar 20 17:30 startpar
    drwxr-xr-x  8 root root  4096 Mar 20 17:31 systemd
    drwxr-xr-x 15 root root  4096 Sep 17  2014 terminfo
    drwxr-xr-x  4 root root  4096 Mar 20 17:31 udev
    drwxr-xr-x  2 root root  4096 Mar 20 17:30 xtables

However, I do have a /sys/firmware directory:

    max@redbox:~$ ls -l /sys/firmware
    total 0
    drwxr-xr-x 4 root root     0 Mar 29 08:05 devicetree
    -r-------- 1 root root 61568 Mar 29 21:15 fdt

Is that of any relevance?

maxwellhadley, 2016-03-30:
One more thing - does octalbonescript still call out to any executables, like the device tree compiler, that might cause these problems if not installed? Remember I started from the console image (which is very lightweight) and added things as needed. I may have not known about something that was needed: although I did install package device-tree-compiler.

maxwellhadley, 2016-03-30:
I have found the problem. It was necessary to install the device-tree-overlays package on the BBB, as it appears to not be included in the stock console flasher BBB-eMMC-flasher-debian-8.3-console-armhf-2016-03-20-2gb.img.

It doesn't appear to be in a standard repository, but I installed it directly from GitHub:

    git clone https://github.com/beagleboard/bb.org-overlays.git
    cd bb.org-overlays
    ./install.sh

octalbonescript is now loading properly. However, it would be nice if Octalbonescript gave an appropriate error message (saying what needs to be done) if the /lib/firmware directory is not found.