theoctal/octalbonescript #42: Problem with the PWM

bAnj0, 2015-07-23:
Hi,
when I try to use the PWM I have this problem on cloud9:

There was an error writing analog value: Fail to update PWM duty: ENOENT, open 'undefined/duty ns'

Suggestions?

adityapatadia, 2015-07-23:
Please give your code here. If possible also output of  ```DEBUG=bone node yourscript.js```

bAnj0, 2015-07-23:
Here the code:

var b = require('octalbonescript');

var motorSpeed = 'P9_21';
var enM1 = 'P9_15';
var M1 = 'P9_16';

b.pinMode(motorSpeed, b.ANALOG_OUTPUT, function(err1) {
  if (err1) {
    console.error(err1.message); 
    return;
 }
});
  
b.pinModeSync(M1, b.OUTPUT);
b.digitalWriteSync(M1, b.LOW);

b.pinModeSync(enM1, b.OUTPUT);
b.digitalWriteSync(enM1, b.HIGH);

b.analogWrite(motorSpeed, 0.5, 2000, function(err2) {
      if (err2) {
        console.error(err2.message); //output any error
        return;
      }
});

ps: where is the output of the debug?

adityapatadia, 2015-07-23:
It will be there in the terminal form where you run your script. Rather than running script from cloud9, I suggest you ssh into bone and run the script by command I provided above.

adityapatadia, 2015-07-23:
The problem is that first pinMode on ```motorSpeed``` is not yet complete and you called ```analogWrite``` on it. Please use ```pinModeSync``` there too, or write your code in callback function of ```pinMode```

closed 2015-07-23 by adityapatadia