theoctal/octalbonescript #45: Documentation about running OBS under OSX

lewisdawson, 2015-08-18:
The README makes reference to the (potential) fact that OBS can be run under OSX in simulator mode:

> This fork is created to make bonescript more feature rich, faster, fix bugs and make it work in simulator mode __under Mac OSX and Linux.__

After digging around a bit, I couldn't find any additional information about this. No tutorials, no documentation, nothing about how to get this to work under OSX. I get the following error when I try to run OBS under OSX (I'm running node v0.10.38):

```<project>/io-interface/node_modules/octalbonescript/node_modules/epoll/node_modules/bindings/bindings.js:83
        throw e
              ^
Error: Symbol epoll_module not found.
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
    at Module.require (module.js:364:17)
    at require (module.js:380:17)
    at bindings (<project>/io-interface/node_modules/octalbonescript/node_modules/epoll/node_modules/bindings/bindings.js:76:44)
    at Object.<anonymous> (<project>/io-interface/node_modules/octalbonescript/node_modules/epoll/epoll.js:1:99)
    at Module._compile (module.js:456:26)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
```

If OBS can in fact be run under OSX, it'd be really awesome to have more information on how to make that work.

lewisdawson, 2015-08-27:
Any information the team can provide on this? Thanks!

-Lewis

adityapatadia, 2015-08-27:
Hi,
When I made OBS, I made is compatible with OS X in simulator mode. That means you can use it in projects and run the code. It will call all pseudo functions during execution. In recent version of OBS, I have updated the epoll library and I did not test it on OS X. 

I will test it this weekend and update OBS so it can again run under OS X in simulator mode.

Hope it helps.

adityapatadia, 2015-09-07:
I asked epoll creator about this issue. I will update when he reverts. If epoll does not work with OS X, I will make it optional and then you will be able to run in simulator mode.

lewisdawson, 2015-09-09:
Thanks for the update @adityapatadia. Let me know what route you decide to take.

lewisdawson, 2015-09-18:
@adityapatadia Any word on this?

adityapatadia, 2015-09-18:
The epoll question I asked did not have any response. I will make it optional tonight. Upgrading to new version will fix it.

fivdi, 2015-10-22:
The epoll question asked was answered on the day it was asked https://github.com/fivdi/epoll/issues/12#issuecomment-138341563

adityapatadia, 2015-12-12:
Ok my bad, I missed it. I am going to make it optional by this weekend.

closed 2015-12-12 by adityapatadia