theoctal/octalbonescript #47: Error reading directory

emersonqueiroz, 2015-11-06:
Hello everyone,

I using the octalbonescript for a while, in a new project i getting two strange behaviours

I getting the bellow output, that is not affecting the sensor read, but i does not know what means:
```
Starting sensor read
Error reading directory /sys/devices/ocp.3 < STRANGE ERROR
Reading left sensor: 0.8911111111111111
Reading left sensor: 0.8888888888888888
```

The code:
```
console.log("Starting sensor read");
readLDR("left", ldr1, cb); << WHEN I CALL THIS FUNCTION, THE OUTPUT ABOVE APPEARS

var readLDR = function(side, ldr, cb) {
    if (!sensorReading) {
        return;
    }
    b.analogRead(ldr, function(err, data) {
        if (err) {
            console.log(">>>>>> ERROR READING SENSOR " + side);
            console.log(err);
            readLDR(side, ldr, CB);
            return;
        }
        console.log("Reading " + side + " sensor: "+ data);
        // SOME VALUES CHECK HERE
        readLDR(side, ldr, cb);
    }); // end read
}
```

I'm constantly reading a sensor through analogRead, but after a few hours it starts to display the following error:
```
{ [VError: analogRead error: EAGAIN, read]
  jse_shortmsg: 'analogRead error',
  jse_summary: 'analogRead error: EAGAIN, read',
  jse_cause: { [Error: EAGAIN, read] errno: 4, code: 'EAGAIN' },
  message: 'analogRead error: EAGAIN, read' }
```

I running the lastest version of octal with Debian

Thank you and sorry for the english :P

adityapatadia, 2015-11-17:
Hi,
I will take a look at the issues in this week. I will get back soon.

adityapatadia, 2015-12-13:
Which kernel do you use? Can you post output of ```uname -a``` command?

adityapatadia, 2015-12-25:
Closing this issue. If you face again, please ask to reopen.

closed 2015-12-25 by adityapatadia