theoctal/octalbonescript #48: Confusion on debian version?

ril3y, 2015-12-08:
I have been trying to get this install via npm for awhile but it always fails.  What beaglebone image works "out of box" with octalbonescript?

adityapatadia, 2015-12-08:
Hi which debian image do you use? I see they updated debian images for beaglebone. I have set a schedule. I will clear most problems around this library by this Sunday including this one. These are low hanging fruits which I will clear for sure.

In past I promised but could not spare enough time. This time we get all issues solved. Any PRs are welcome.

adityapatadia, 2015-12-08:
I last checked with this image. Debian 7.8 (BeagleBone, BeagleBone Black - 4GB SD) 2015-03-01. I see that latest one is Debian 8.2. I am upgrading my own board and will upgrade OBS in a few days.

ril3y, 2015-12-08:
Specifically?

http://debian.beagleboard.org/images/bone-debian-7.8-lxde-4gb-armhf-2015-03-01-4gb.img.xz

I have tried 8.2 and 7.9 and both fail.  Also  I have tried with the "built
in node" and upgraded node too.  There is no documentation on what debian
version works with OBS.  I think that it should be linked to and stated
specifically so people don't fall down the " this  doesn't work" kind of
issue.

I will try 7.8.

On a side note it looks like for 8.2 the paths are wrong.  I have not had
the time to try to fix it, nor do I fully understand if a  pull request
would work.  Meaning, its almost like OBS needs some sort of release
checking to update paths and other "gotchas"

Thanks!

adityapatadia, 2015-12-08:
Ok, I will update the readme and wiki about supported version. It happened because I assumed that kernels wont change that fast. 

You can expect new version resolving this by this Sunday.

ril3y, 2015-12-08:
Thanks!  That would be great.  I have a project that I am short on time and
really really am not that great on C++!  So I would love to use OBS vs lots
of compiled code!

Thanks again Aditya!

ril3y, 2015-12-08:
I just tired the link above that I posted but it fails as well.  FYI.  Thanks!

adityapatadia, 2015-12-08:
It’s this link: Debian 7.8 (BeagleBone, BeagleBone Black - 4GB SD) 2015-03-01 <http://debian.beagleboard.org/images/bone-debian-7.8-lxde-4gb-armhf-2015-03-01-4gb.img.xz>
Regards,
Aditya Patadia

ril3y, 2015-12-09:
@adityapatadia I tried out the version you listed tonight.  Fresh install, did nothing to it and tried  a sudo npm install octalbonescript but this fails also.

Do you have to do something special?  I upgrade node to v.12 to see if that would help.  But it continues to fail.  Any ideas?

adityapatadia, 2015-12-09:
Please post errors you get.

ril3y, 2015-12-10:
I did this from a fresh install from the image you linked in above.  The only thing I did was change the password on the debian user and then sudo -s to root.

root@beaglebone:/home/debian# npm install -g octalbonescript

|
> i2c@0.1.8 install /usr/local/lib/node_modules/octalbonescript/node_modules/i2c
> node-gyp rebuild

make: Entering directory `/usr/local/lib/node_modules/octalbonescript/node_modules/i2c/build'
  CXX(target) Release/obj.target/i2c/src/i2c.o
  SOLINK_MODULE(target) Release/obj.target/i2c.node
  SOLINK_MODULE(target) Release/obj.target/i2c.node: Finished
  COPY Release/i2c.node
make: Leaving directory `/usr/local/lib/node_modules/octalbonescript/node_modules/i2c/build'

> epoll@0.1.16 install /usr/local/lib/node_modules/octalbonescript/node_modules/epoll
> node-gyp rebuild

make: Entering directory `/usr/local/lib/node_modules/octalbonescript/node_modules/epoll/build'
  CXX(target) Release/obj.target/epoll/src/epoll.o
In file included from ../src/epoll.cc:15:0:
../node_modules/nan/nan.h:328:47: error: ‘REPLACE_INVALID_UTF8’ is not a member of ‘v8::String’
cc1plus: warning: unrecognized command line option "-Wno-unused-local-typedefs" [enabled by default]
make: *** [Release/obj.target/epoll/src/epoll.o] Error 1
make: Leaving directory `/usr/local/lib/node_modules/octalbonescript/node_modules/epoll/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/share/node-gyp/lib/build.js:267:23)
gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:809:12)
gyp ERR! System Linux 3.8.13-bone70
gyp ERR! command "nodejs" "/usr/bin/node-gyp" "rebuild"
gyp ERR! cwd /usr/local/lib/node_modules/octalbonescript/node_modules/epoll
gyp ERR! node -v v0.10.29
gyp ERR! node-gyp -v v0.12.2
gyp ERR! not ok 
npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian
 
npm ERR! epoll@0.1.16 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the epoll@0.1.16 install script.
npm ERR! This is most likely a problem with the epoll package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls epoll
npm ERR! There is likely additional logging output above.

npm ERR! System Linux 3.8.13-bone70
npm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "install" "-g" "octalbonescript"
npm ERR! cwd /home/debian
npm ERR! node -v v0.10.29
npm ERR! npm -v 1.4.21
npm ERR! code ELIFECYCLE
npm ERR! 
npm ERR! Additional logging details can be found in:
npm ERR!     /home/debian/npm-debug.log
npm ERR! not ok code 0

This is the error code.

adityapatadia, 2015-12-12:
I asked the question on epoll module. Let's wait for them to respond. I will post a fix when I find one.

adityapatadia, 2015-12-12:
I am getting another issue on my BeagleBone after upgrading it to Debian 8.2. I have posted question on epoll module. Hope to get it too resolved soon.

adityapatadia, 2015-12-12:
There is also some issue about how permissions are handled in npm. I am updating readme to recommend installing octalbonescript locally on per project basis. Check this issue for more information: https://github.com/fivdi/epoll/issues/17

closed 2015-12-12 by adityapatadia

fivdi, 2015-12-12:
@adityapatadia above you make the following comment:

>This is a known issue for epoll module. Please refer to https://github.com/fivdi/onoff/wiki/Node.js-v0.10.29-and-native-addons-on-the-BeagleBone-Black

I'd just like to note that this comment as it's totally untrue :)
It's not an epoll issue and has nothing to do with the document linked to.

In a later comment you do however link to https://github.com/fivdi/epoll/issues/17 which describes what the problem really is.

adityapatadia, 2015-12-12:
@ril3y mentioned in his error log that he is getting ‘REPLACE_INVALID_UTF8’ error. That issue is related to node version 0.10.29. We can also see from his error log that he is running 0.10.29 version of node. 

I also updated my comment to reflect that issue is with nodejs and not only in epoll module. I also added my last comment to educate any user as these issues seems very close from error logs.

fivdi, 2015-12-12:
Your quite correct. Sorry for adding to the confusion. I'm going to delete my last comment and I'll delete this comment in a few minutes.

ril3y, 2015-12-13:
@adityapatadia have you got octalbonescript to build with the image you linked in to me above?  Without using -g?  

thanks

adityapatadia, 2015-12-13:
Yes, I could install it on a per project basis by doing ```npm install octalbonescript``` in given project folder.

bigmonkeyboy, 2015-12-25:
I find npm v2 much better at handling some of these... especially rebuilding things like newer versions of serialport etc.

    sudo npm i -g npm@2.x

usually get you there.