theoctal/octalbonescript #50: Issue with getting octalbonescript going with other cape

psiphi75, 2015-12-19:
Hi,

I am trying to get OBS going with another cape.  The other cape is the ADAFRUIT-UART2 cape, that enables the serial port.  I require the serial port for a GPRS modem.

I am running Linux 3.8.13, on Debian 7.9 with NodeJS 0.10.38, OBS v1.0.0 as root.  When I run OBS with just the code below:

```javascript
    var b = require('octalbonescript');
```

I get the following error message:

```
  bone is_ocp() = /sys/devices/ocp.3 +0ms
  bone is_cape_universal() = null +30ms
  bone Loading Universal Cape interface... +14ms
  bone create_dt_sync(OBS_UNIV,,) +20ms
  bone onDTBOExists() +12ms
  bone load_dt_sync(OBS_UNIV) +11ms
  bone onFindCapeMgr: path = /sys/devices/bone_capemgr.9 +11ms
  bone onReadSlots: index = -1, readAttempts = 1 +21ms
  bone Writing OBS_UNIV to /sys/devices/bone_capemgr.9/slots +6ms
VError: Write to CapeMgr slots failed: EEXIST, file already exists
        at onWriteSlots (/var/lib/cloud9/node_modules/octalbonescript/lib/bone.js:253:23)
        at onReadSlots (/var/lib/cloud9/node_modules/octalbonescript/lib/bone.js:244:17)
        at readSlots (/var/lib/cloud9/node_modules/octalbonescript/lib/bone.js:224:13)
        at onFindCapeMgr (/var/lib/cloud9/node_modules/octalbonescript/lib/bone.js:214:13)
        at Object.module.exports.load_dt_sync (/var/lib/cloud9/node_modules/octalbonescript/lib/bone.js:204:9)
        at onDTBOExists (/var/lib/cloud9/node_modules/octalbonescript/lib/bone.js:189:38)
        at onDTBOExistsTest (/var/lib/cloud9/node_modules/octalbonescript/lib/bone.js:156:17)
        at Object.module.exports.create_dt_sync (/var/lib/cloud9/node_modules/octalbonescript/lib/bone.js:151:13)
        at Object.<anonymous> (/var/lib/cloud9/node_modules/octalbonescript/index.js:30:14)
        at Module._compile (module.js:456:26)
```    

Do you have any ideas on how to get UART2 (or others) to work together with OBS?

adityapatadia, 2015-12-24:
Please be ready to upgrade your board to debian 8.2. I will release new OBS v1.1.0 by this Sunday. There will be functionality in new version to address this issue.

adityapatadia, 2015-12-26:
Please refer to https://github.com/theoctal/octalbonescript/wiki/Load-Unload-Cape to unload universal cape and load desired one.

closed 2015-12-26 by adityapatadia

psiphi75, 2016-01-07:
I still have the same issue.  I have upgraded to Linux 4.1.13-ti-r38, Debian 8.2 with NodeJS 0.10.41, OBS v1.1.1 as root.

Capes can be loaded/unloaded just fine.  But I cannot get UART2 working at the same time as the PWM (P8_13).  I will raise a new issue that OBS does not work with the MIKROBUS cape.