theoctal/octalbonescript #51: missing callback causes crash

bigmonkeyboy, 2015-12-22:
I know the new style is to use callbacks for everything... but... if the callback is missing then the script errors out...

eg calling digital write with just a pin and value causes

    Uncaught Exception:
    TypeError: undefined is not a function
    at onWriteGPIO (/root/node_modules/octalbonescript/lib/digital.js:185:17)
    ....

which comes from a `callback(null);`  where of course callback itself is not defined...
Ideally I think you should allow the user not to have to define a callback... if they don't want to.

adityapatadia, 2015-12-22:
This I will fix this in new version. It’s indeed a bug. Please list which functions throw this error.

bigmonkeyboy, 2015-12-22:
Hi - a quick `grep -rn "    callback" ` in the lib directory would seem to indicate that there are quite a few places - some of which may be handled by a surrounding if block... but many are not.

adityapatadia, 2015-12-22:
:) sure. I will get them fixed.

closed 2015-12-25 by adityapatadia

bigmonkeyboy, 2015-12-25:
Yes - much better... Thanks !

commit bd44b232 (2015-12-25): fix #51