theoctal/octalbonescript #53: Write to CapeMgr slots failed: ENOENT, no such file or directory

Kadalia, 2016-01-05:
Hello,
got the error "Write to CapeMgr slots failed: ENOENT, no such file or directory"  with a simple script.

I dont know if it's a problem, when i do ls /lib/firmware/
i get : dra7-ipu2-fw.xem4  dra7xx-m4-ipu2.xem4  vpdma-1b8.bin

Beagle bone installed with debian
uname -a : Linux beaglebone 4.1.13-ti-r38 #1 SMP PREEMPT Thu Dec 31 23:38:01 UTC 2015 armv7l GNU/Linux

Thank you !
Philippe

Kadalia, 2016-01-05:
debug output :

  bone is_ocp() = /sys/devices/platform/ocp +0ms
  bone find_sysfsFile(cape-universal,/sys/devices/platform/ocp,ocp:cape-universal) +23ms
  bone is_cape_universal() = null +10ms
  bone Loading Universal Cape interface... +2ms
  bone load_dt_sync(cape-universaln) +4ms
  bone onFindCapeMgr: path = /sys/devices/platform/bone_capemgr +2ms
  bone cape-universaln +5ms
  bone onReadSlots: index = -1, readAttempts = 1 +2ms
  bone Writing cape-universaln to /sys/devices/platform/bone_capemgr/slots +1ms

Kadalia, 2016-01-05:
bone.js       fs.writeFileSync(slotsFile, name, 'ascii');  (near line 200)
rises an error ENOENT, no such file or directory
but the file exists !!!
  debug(fs.existsSync(slotsFile));  return true !!!!!!!!
if i comment onWriteSlots(err2);
seems to work !
heu ... I do not understand !!!

adityapatadia, 2016-01-06:
What is your original function call? Can you paste code here?

adityapatadia, 2016-01-06:
Btw, the error is that it found slots file but it is not able to find .dtbo file for cape-universaln inside /lib/firmware folder. I suggest u re-install latest debian from link given in readme.

adityapatadia, 2016-01-08:
Was this solved?

adityapatadia, 2016-01-21:
Closing it for now, I will re-open if the reporter requests.

closed 2016-01-21 by adityapatadia