theoctal/octalbonescript #54: Get "Error waiting for CapeMgr slot to load", but cape loads

psiphi75, 2016-01-07:
When I run the following code:

```javascript
var b = require('octalbonescript');
b.unloadCape('cape-univ-hdmi');
b.unloadCape('BB-ADC');
b.unloadCape('cape-universaln');

b.loadCape('BB-MIKROBUS-01:00A1');
```

I get the following error message:
```
  bone load_dt_sync(BB-MIKROBUS-01:00A1) +7s
  bone onFindCapeMgr: path = /sys/devices/platform/bone_capemgr +3ms
  bone onReadSlots: index = -1, readAttempts = 1 +5ms
  bone Writing BB-MIKROBUS-01:00A1 to /sys/devices/platform/bone_capemgr/slots +3ms
  bone onReadSlots: index = -1, readAttempts = 2 +256ms
VError: Error waiting for CapeMgr slot to load
    at onReadSlots (/root/node_modules/octalbonescript/lib/bone.js:199:23)
    at readSlots (/root/node_modules/octalbonescript/lib/bone.js:176:13)
    at onWriteSlots (/root/node_modules/octalbonescript/lib/bone.js:207:13)
    at onReadSlots (/root/node_modules/octalbonescript/lib/bone.js:196:17)
    at readSlots (/root/node_modules/octalbonescript/lib/bone.js:176:13)
    at Object.module.exports.load_dt_sync (/root/node_modules/octalbonescript/lib/bone.js:167:9)
    at Object.f.loadCape (/root/node_modules/octalbonescript/index.js:56:15)
    at repl:1:3
    at REPLServer.self.eval (repl.js:110:21)
    at repl.js:249:20
```

However, the cape loaded fine:

```
root@beaglebone:~# cat /sys/devices/platform/bone_capemgr/slots 
 0: PF----  -1 
 1: PF----  -1 
 2: PF----  -1 
 3: PF----  -1 
19: P-O-L-   1 Override Board Name,00A1,Override Manuf,BB-MIKROBUS-01
```

Note the MIKROBUS-01 cape can be compiled using these instructions: http://www.libstock.com/projects/view/1525/remote-weather-station

adityapatadia, 2016-01-21:
Is this still an issue?

psiphi75, 2016-01-23:
Technically this is still an issue.  But I have found that I am able to achieve what I need with the default capes.

Thanks for your support.

adityapatadia, 2016-01-23:
Ohk, I will solve this issue and then close it. Expect a code push for this within a day or two.

adityapatadia, 2016-03-22:
Is there any other compiled cap already available which I can use? I don't want to compile and test everything on my own if there is some easier ready-made cape is available.

psiphi75, 2016-03-22:
I have found that the universal cape works just fine.  So I don't need to use the BB-MIKROBUS-01.  From my point this issue can be closed.

closed 2016-03-22 by psiphi75