theoctal/octalbonescript #55: MikroBus Cape does not work with Octalbonescript

psiphi75, 2016-01-07:
I am trying to get OBS (octalbonescript) working together with the MikroBus Cape (http://www.mikroe.com/click/mikrobus-cape/).  There is a Device Tree Overlay (DTO) for the MikroBus cape can be compiled using the instructions on this page: http://www.libstock.com/projects/view/1525/remote-weather-station

I use the following code:
```javascript
var b = require('octalbonescript');

// Unload the stuff I don't want
b.unloadCape('cape-univ-hdmi');
b.unloadCape('BB-ADC');
b.unloadCape('cape-universaln');

// Load the stuff I want
b.loadCape('BB-MIKROBUS-01:00A1');

// Get the PWM going
var SERVO = 'P8_13';
b.pinMode(SERVO, b.OUTPUT, function (err, result) { console.log(err, result); });

```

It gives me the following error:
```
VError: The pin P8_13 is not availble to write. Please make sure it is not used by another cape.
    at Object.module.exports.getpin (/root/node_modules/octalbonescript/lib/bone.js:46:27)
    at Object.f.pinMode (/root/node_modules/octalbonescript/index.js:165:20)
    at repl:1:3
    at REPLServer.self.eval (repl.js:110:21)
    at repl.js:249:20
    at REPLServer.self.eval (repl.js:122:7)
    at Interface.<anonymous> (repl.js:239:12)
    at Interface.emit (events.js:95:17)
    at Interface._onLine (readline.js:203:10)
    at Interface._line (readline.js:532:8)
```

adityapatadia, 2016-01-08:
I will take a look on both errors this weekend and solve it.

adityapatadia, 2016-01-12:
Can you post some documentation about MicroBus Cape? I am unaware of what pins it will use. I will need to take a look into its dtb file.

adityapatadia, 2016-01-12:
I got the dts file, I will take a look and let you know next week.

adityapatadia, 2016-01-18:
Hi, why would you like to unload universal cape and load microbus cape before writing to P8_13? Isn't it something you can do with universal cape itself? I am trying to understand microbus cape so your help will be appreciated.

psiphi75, 2016-01-18:
It appears I can't load both the universal and MikroBus capes at the same time.  But I would like to load the MikroBus cape because that is how my devices are configured.  The universal cape does not work for me, I cannot load the UARTs with the universal cape.

I digress here a little, but ultimately the following solution would work for me:
The end result I would like is to have many devices connected using a custom DTO.  I believe OBS should be able to create a custom DTO (although I'm not saying it's easy).  I could imagine a function is created that assigns a pin to a GPIO/PWM/UART, etc.

For example:
```javascript
var obs = require('octalbonescript');
obs.assignPin('P9_14', 'PWM1A');
obs.assignPin('P9_16', 'PWM1B');
obs.assignPin('P9_11', 'UART4_RX');
obs.assignPin('P9_13', 'UART4_TX');
obs.assignPin('P9_15', 'GPIO_48');

// This would apply the assign pins by compiling the DTO and loading the cape
obs.applyAssignments();

b.pinMode('P9_14', b.OUTPUT, function (err, result) { console.log(err, result); });
```

This is a hypothetical example, but it would allow you to fully customise which pins are assigned which functionality.  My knowledge of OBS, DTOs and the BeagleBone Black is limited, so this may not be possible.

adityapatadia, 2016-01-19:
Hi, what you explain is already possible with OBS universal cape. The rationale behind developing universal cape was to eliminate any need to load any other cape.

Let me give real world example from your hypothetical one.

```javascript

var obs = require('octalbonescript');
obs.pinMode("P9_14", obs.ANALOG_OUTPUT); //makes the pin PWM
obs.pinMode("P9_15", obs.OUTPUT); // makes the pin GPIO
obs.serial.open('/dev/tty01'); // makes the pins UARTs
```

For more information on serial ports, refer this: https://github.com/theoctal/octalbonescript/wiki/Serial-Port

Let me know if you need further help.

adityapatadia, 2016-01-21:
Could this solve issue for you?

psiphi75, 2016-01-22:
Yes, it did.  Now I get how OBS works, it's great!  But I had to use slightly different code.  Because I wanted to enable the serial port (not use it), I had to use the `enable` function.  Here is the code that worked for me:

```javascript
var obs = require('octalbonescript');
obs.pinMode("P9_14", obs.ANALOG_OUTPUT, function () {});
obs.pinMode("P9_15", obs.OUTPUT, function () {});
obs.serial.enable('/dev/ttyO2', function() {
    console.log('enabled serial');
});
```

Thanks a lot for your support.

adityapatadia, 2016-01-23:
I am glad that it worked. Feel free to ping in future in case of any issues.

closed 2016-01-23 by adityapatadia