theoctal/octalbonescript #56: Octalbonescript installed with Node-red - Error: Cannot find module 'octalbonescript'

michelmimi, 2016-01-28:
Hello,
As described in the "nodered.org"  guidelines, I installed octalbonescript within node-red environnement.
and uncommented the line related to octalbonescript in the file settings.js in "functionGlobalContext"

> functionGlobalContext: {
>     // os:require('os'),
>     octalbonescript:require('octalbonescript'),
>     // jfive:require("johnny-five"),
>     // j5board:require("johnny-five").Board({repl:false})
> },

Unhappily, each time I want to start node-red flow, I get the following error message:

> root@arm:~/.node-red# node-red-pi -v
> Error loading settings file: /root/.node-red/settings.js
> Error: Cannot find module 'octalbonescript'

When the line is commented again, node-red is starting correctly but errors appears as soon as I use a Beaglebone Black input in node-red graphical environnement. The errors appear below.

> root@arm:~/.node-red# node-red-pi -v
> 
> 
> Welcome to Node-RED
> ===================
> 
> 28 Jan 10:13:43 - [info] Node-RED version: v0.13.1
> 28 Jan 10:13:43 - [info] Node.js  version: v0.10.25
> 28 Jan 10:13:43 - [info] Loading palette nodes
> 28 Jan 10:13:55 - [warn] ------------------------------------------
> 28 Jan 10:13:55 - [warn] [rpi-gpio] Info : Ignoring Raspberry Pi specific node
> 28 Jan 10:13:55 - [warn] ------------------------------------------
> 28 Jan 10:13:55 - [info] Settings file  : /root/.node-red/settings.js
> 28 Jan 10:13:55 - [info] User directory : /root/.node-red
> 28 Jan 10:13:55 - [info] Flows file : /root/.node-red/flows_arm.json
> 28 Jan 10:13:55 - [info] Creating new flow file
> 28 Jan 10:13:55 - [info] Starting flows
> 28 Jan 10:13:55 - [info] Started flows
> 28 Jan 10:13:55 - [info] Server now running at http://127.0.0.1:1880/
> 28 Jan 10:18:36 - [info] Stopping modified flows
> 28 Jan 10:18:36 - [info] Stopped modified flows
> 28 Jan 10:18:36 - [info] Starting modified flows
> 28 Jan 10:18:36 - [info] Started modified flows
> 28 Jan 10:22:12 - [info] Stopping modified flows
> 28 Jan 10:22:12 - [info] Stopped modified flows
> 28 Jan 10:22:12 - [info] Starting modified flows
> 28 Jan 10:22:12 - [info] Started modified flows
> 28 Jan 10:23:53 - [info] Stopping modified flows
> 28 Jan 10:23:53 - [info] Stopped modified flows
> 28 Jan 10:23:53 - [info] Starting modified flows
> 28 Jan 10:23:53 - [info] Started modified flows
> 28 Jan 10:23:53 - [red] Uncaught Exception:
> 28 Jan 10:23:53 - TypeError: Cannot read property 'value' of null
>     at /usr/local/lib/node_modules/node-red-node-beaglebone/145-BBB-hardware.js:249:33
>     at onExport (/usr/local/lib/node_modules/node-red-node-beaglebone/node_modules/octalbonescript/index.js:255:47)
>     at onGPIODirectionSet (/usr/local/lib/node_modules/node-red-node-beaglebone/node_modules/octalbonescript/lib/digital.js:134:51)
>     at /usr/local/lib/node_modules/node-red/node_modules/fs-extra/node_modules/graceful-fs/graceful-fs.js:42:10
>     at Object.oncomplete (fs.js:107:15)
> root@arm:~/.node-red# 

Octalbonescript module was installed as root with the following command in .node-red/ directory:

> npm install -g --unsafe-perm octalbonescript

So I have no idea why this happens. Do you have some suggestion to solve this problem ?

Kind regards

Michel

adityapatadia, 2016-03-22:
I am not sure why are you having the issue. I suggest you put octalbonescript as dependency in package.json file and then do 'npm install' in that directory to install octalbonescript. Do you face issues with other node modules as well?

adityapatadia, 2016-04-02:
Closing this for now. Let me know if issue still persists. I will reopen.

closed 2016-04-02 by adityapatadia