theoctal/octalbonescript #57: AnalogWriteSync?

kory44, 2016-02-07:
Is there a way to get synchronous AnalogWrite? I don't see it in the docs? Here is the code i want to migrate from stock bonescript:

`function GrowVent(in1,in2,enable){
	var self = this;
	self.vent_speed = 0.39; //voltage, controls speed, a calibrated valuw
	self.vent_wait = 900; //milliseconds, a calibrated value
	self.in1 = in1;
	self.in2 = in2;
	self.state = 0;
	self.enable = enable;
	bone.pinMode(self.in1, 'out');
	bone.pinMode(self.in2, 'out');
	//bone.pinMode(self.enable, bone.ANALOG_OUTPUT);
	self.Open = function(){
		bone.digitalWrite(self.in1,1);
		bone.digitalWrite(self.in2,0);
		bone.analogWrite(self.enable,self.vent_speed);
		setTimeout(bone.analogWrite, self.vent_wait, self.enable, 0.0);
		self.state = 1;
	}
	self.Close = function(){
		bone.digitalWrite(self.in1,0);
		bone.digitalWrite(self.in2,1);
		bone.analogWrite(self.enable,self.vent_speed);
		setTimeout(bone.analogWrite, self.vent_wait, self.enable, 0.0);
		self.state = 0;
	}
}

Vent_main = new GrowVent("P8_14", "P8_7", "P9_29");`

adityapatadia, 2016-02-09:
Unfortunately, the way linux kernel is made, it is not much possible to write analogWriteSync function. Instead,I would write code like this. Let me know if you face any issues.

```javascript

var GrowVent = {
    vent_speed : 0.39, //voltage, controls speed, a calibrated valuw
    vent_wait : 900, //milliseconds, a calibrated value
    in1 : null,
    in2 : null,
    state : 0,
    enable : null,
    init : function(in1, in2, enable){
         this.in1 = in1;
         this.in2 = in2;
         this.enable = enable;
         bone.pinMode(in1, 'out');
         bone.pinMode(in2, 'out');
    },
    open : function(){
        var self = this;
        bone.digitalWriteSync(self.in1,1);
        bone.digitalWriteSync(self.in2,0);
        bone.analogWrite(self.enable, self.vent_speed, 2000, function(err){ // the callback function will be called when the pin is analog written. It will then put setTimeout.
            setTimeout(function(){
                  bone.analogWrite(self.enable, 0);
            ,self.wait);
           self.state = 1;
        });
    },
    close : function(){
        var self = this;
        bone.digitalWriteSync(self.in1,0);
        bone.digitalWriteSync(self.in2,1);
        bone.analogWrite(self.enable, self.vent_speed, 2000, function(err){
            setTimeout(function(){
                  bone.analogWrite(self.enable, 0);
            ,self.wait);
           self.state = 1;
        });
    }
}    
Vent_main = new GrowVent("P8_14", "P8_7", "P9_29");

GrowVent.init("P8_14","P8_7","P9_29"); // this is async call,it will take some time. You can replace pinMode with pinModeSync in init function to make it sync.
GrowVent.open(); // this will also be async call. Let me know if you want to make it sync. I will write code for it.
GrowVent.close(); // this will be async call, let me know if you want to make it sync.

```

kory44, 2016-02-09:
For others that may see this, could you show synchronous open and close methods ?

kory44, 2016-03-07:
What could cause this error?

function GrowVent(in1,in2,enable){
	var self = this;
	self.vent_speed = 0.2;//0.39; //voltage, controls speed, a calibrated value
	self.vent_wait = 200;//900; //milliseconds, a calibrated value
	self.in1 = in1;
	self.in2 = in2;
	self.state = 0;
	self.enable = enable;
	bone.pinModeSync(self.in1, 'out');
	bone.pinModeSync(self.in2, 'out');
	bone.pinModeSync(self.enable, bone.ANALOG_OUTPUT);
	self.Open = function(){
		bone.digitalWriteSync(self.in1,0);
		bone.digitalWriteSync(self.in2,1);
		bone.analogWrite(self.enable,self.vent_speed,2000.0,function(){
			setTimeout(bone.analogWrite, self.vent_wait, self.enable, 0.0);
			self.state = 1;
	});}
	self.Close = function(){
		bone.digitalWriteSync(self.in1,1);
		bone.digitalWriteSync(self.in2,0);
		bone.analogWrite(self.enable,self.vent_speed,2000.0,function(){
			setTimeout(bone.analogWrite, self.vent_wait, self.enable, 0.0);
			self.state = 0;
	});}
}

> Vent_root = new GrowVent("P8_12", "P8_11","P9_14");
VError: The pin P8_12 is not availble to write. Please make sure it is not used by another cape.
    at Object.module.exports.getpin (/var/lib/cloud9/node_modules/octalbonescript/lib/bone.js:47:27)
    at Object.f.pinModeSync (/var/lib/cloud9/node_modules/octalbonescript/index.js:248:16)
    at new GrowVent (repl:9:6)
    at repl:1:13
    at REPLServer.self.eval (repl.js:110:21)
    at repl.js:249:20
    at REPLServer.self.eval (repl.js:122:7)
    at Interface.<anonymous> (repl.js:239:12)
    at Interface.emit (events.js:95:17)
    at Interface._onLine (readline.js:203:10)

adityapatadia, 2016-04-02:
That means the pin is used by some other cape. Did you load any other cape? If not, please try some other pin. Is it that you must use P8_12?

kory44, 2016-04-02:
I ran this code directly from boot. is something built in that is using a
cape? I'm trying to use every pwm so I need this pin to work.

adityapatadia, 2016-04-03:
Let me check P8_12 and get back to you.

kory44, 2016-04-03:
wait,I never said P8_12... My trouble is with P9_31 and P9_14.

Thanks,

Kory De Angelo
k.deangelo44@gmail.com
(719) 291-1954

adityapatadia, 2016-04-04:
Your error output says P8_12 is not available to write. Anyway let me test all 3 pins. You can expect response in 3-4 days.

Andrewiski, 2016-06-03:
@kory44  Assuming you are using debian with cape manager support, you may want to check to see if the universal cape is loaded. 

a check of ‘cat /sys/devices/platform/bone_capemgr/slots’ output is
0: PF---- -1
1: PF---- -1
2: PF---- -1
3: PF---- -1

if there is no 4: with a universal cape that may be the issue.

adityapatadia, 2017-06-07:
Please let us know if it was resolved?

adityapatadia, 2017-12-24:
This library is abandoned. Please check notice in README for further information.

closed 2017-12-24 by adityapatadia