theoctal/octalbonescript #58: can't install octal bonescript on debian 3.8

kory44, 2016-02-24:
I saw that it is recommended to use version 1.0.0 on 3.8 kernel. I tried to install it but got a lot of errors. What more info do I need to provide?

root@beaglebone:/var/lib/cloud9/grow-io/realgrow# npm install octalbonescript@1.0.0
\
> i2c@0.1.8 install /var/lib/cloud9/grow-io/realgrow/node_modules/octalbonescript/node_modules/i2c
> node-gyp rebuild

make: Entering directory `/var/lib/cloud9/grow-io/realgrow/node_modules/octalbonescript/node_modules/i2c/build'
  CXX(target) Release/obj.target/i2c/src/i2c.o
  SOLINK_MODULE(target) Release/obj.target/i2c.node
  SOLINK_MODULE(target) Release/obj.target/i2c.node: Finished
  COPY Release/i2c.node
make: Leaving directory `/var/lib/cloud9/grow-io/realgrow/node_modules/octalbonescript/node_modules/i2c/build'

> epoll@0.1.17 install /var/lib/cloud9/grow-io/realgrow/node_modules/octalbonescript/node_modules/epoll
> node-gyp rebuild

make: Entering directory `/var/lib/cloud9/grow-io/realgrow/node_modules/octalbonescript/node_modules/epoll/build'
  CXX(target) Release/obj.target/epoll/src/epoll.o
In file included from ../src/epoll.cc:15:0:
../node_modules/nan/nan.h:328:47: error: ‘REPLACE_INVALID_UTF8’ is not a member of ‘v8::String’
cc1plus: warning: unrecognized command line option "-Wno-unused-local-typedefs" [enabled by default]
make: *** [Release/obj.target/epoll/src/epoll.o] Error 1
make: Leaving directory `/var/lib/cloud9/grow-io/realgrow/node_modules/octalbonescript/node_modules/epoll/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/share/node-gyp/lib/build.js:267:23)
gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:809:12)
gyp ERR! System Linux 3.8.13-bone70
gyp ERR! command "nodejs" "/usr/bin/node-gyp" "rebuild"
gyp ERR! cwd /var/lib/cloud9/grow-io/realgrow/node_modules/octalbonescript/node_modules/epoll
gyp ERR! node -v v0.10.29
gyp ERR! node-gyp -v v0.12.2
gyp ERR! not ok 
npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian
 

> serialport@1.4.10 install /var/lib/cloud9/grow-io/realgrow/node_modules/octalbonescript/node_modules/serialport
> node-pre-gyp install --fallback-to-build

module.js:340
    throw err;
          ^
Error: Cannot find module '/var/lib/cloud9/grow-io/realgrow/node_modules/octalbonescript/node_modules/serialport/node_modules/.bin/node-pre-gyp'
    at Function.Module._resolveFilename (module.js:338:15)
    at Function.Module._load (module.js:280:25)
    at Function.Module.runMain (module.js:497:10)
    at startup (node.js:119:16)
    at node.js:906:3
npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian
 
npm ERR! epoll@0.1.17 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the epoll@0.1.17 install script.
npm ERR! This is most likely a problem with the epoll package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls epoll
npm ERR! There is likely additional logging output above.

npm ERR! System Linux 3.8.13-bone70
npm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "install" "octalbonescript@1.0.0"
npm ERR! cwd /var/lib/cloud9/grow-io/realgrow
npm ERR! node -v v0.10.29
npm ERR! npm -v 1.4.21
npm ERR! code ELIFECYCLE
npm ERR! 
npm ERR! Additional logging details can be found in:
npm ERR!     /var/lib/cloud9/grow-io/realgrow/npm-debug.log
npm ERR! not ok code 0

adityapatadia, 2016-03-06:
Can you try ```npm install octalbonescript@1.0.6``` ?

kory44, 2016-03-07:
it still didn't work, is there a problem with my node version? I know nothing about node...

root@beaglebone:/var/lib/cloud9/grow-io/realgrow/node_modules# ls -a
.  ..
root@beaglebone:/var/lib/cloud9/grow-io/realgrow/node_modules# cd ..
root@beaglebone:/var/lib/cloud9/grow-io/realgrow# npm install octalbonescript@1.0.6
-
> i2c@0.1.8 install /var/lib/cloud9/grow-io/realgrow/node_modules/octalbonescript/node_modules/i2c
> node-gyp rebuild

make: Entering directory `/var/lib/cloud9/grow-io/realgrow/node_modules/octalbonescript/node_modules/i2c/build'
  CXX(target) Release/obj.target/i2c/src/i2c.o
  SOLINK_MODULE(target) Release/obj.target/i2c.node
  SOLINK_MODULE(target) Release/obj.target/i2c.node: Finished
  COPY Release/i2c.node
make: Leaving directory `/var/lib/cloud9/grow-io/realgrow/node_modules/octalbonescript/node_modules/i2c/build'

> epoll@0.1.17 install /var/lib/cloud9/grow-io/realgrow/node_modules/octalbonescript/node_modules/epoll
> node-gyp rebuild

make: Entering directory `/var/lib/cloud9/grow-io/realgrow/node_modules/octalbonescript/node_modules/epoll/build'
  CXX(target) Release/obj.target/epoll/src/epoll.o
In file included from ../src/epoll.cc:15:0:
../node_modules/nan/nan.h:328:47: error: ‘REPLACE_INVALID_UTF8’ is not a member of ‘v8::String’
cc1plus: warning: unrecognized command line option "-Wno-unused-local-typedefs" [enabled by default]
make: *** [Release/obj.target/epoll/src/epoll.o] Error 1
make: Leaving directory `/var/lib/cloud9/grow-io/realgrow/node_modules/octalbonescript/node_modules/epoll/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/share/node-gyp/lib/build.js:267:23)
gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:809:12)
gyp ERR! System Linux 3.8.13-bone70
gyp ERR! command "nodejs" "/usr/bin/node-gyp" "rebuild"
gyp ERR! cwd /var/lib/cloud9/grow-io/realgrow/node_modules/octalbonescript/node_modules/epoll
gyp ERR! node -v v0.10.29
gyp ERR! node-gyp -v v0.12.2
gyp ERR! not ok 
npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian
 
npm ERR! error rolling back Error: ENOTEMPTY, rmdir '/var/lib/cloud9/grow-io/realgrow/node_modules/octalbonescript/node_modules/serialport/node_modules/optimist/node_modules/wordwrap'
npm ERR! error rolling back  octalbonescript@1.0.6 { [Error: ENOTEMPTY, rmdir '/var/lib/cloud9/grow-io/realgrow/node_modules/octalbonescript/node_modules/serialport/node_modules/optimist/node_modules/wordwrap']
npm ERR! error rolling back   errno: 53,
npm ERR! error rolling back   code: 'ENOTEMPTY',
npm ERR! error rolling back   path: '/var/lib/cloud9/grow-io/realgrow/node_modules/octalbonescript/node_modules/serialport/node_modules/optimist/node_modules/wordwrap' }
npm ERR! epoll@0.1.17 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the epoll@0.1.17 install script.
npm ERR! This is most likely a problem with the epoll package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls epoll
npm ERR! There is likely additional logging output above.

npm ERR! System Linux 3.8.13-bone70
npm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "install" "octalbonescript@1.0.6"
npm ERR! cwd /var/lib/cloud9/grow-io/realgrow
npm ERR! node -v v0.10.29
npm ERR! npm -v 1.4.21
npm ERR! code ELIFECYCLE
npm ERR! Error: ENOENT, lstat '/var/lib/cloud9/grow-io/realgrow/node_modules/octalbonescript/node_modules/serialport/node_modules/optimist/node_modules/minimist/index.js'
npm ERR! If you need help, you may report this *entire* log,
npm ERR! including the npm and node versions, at:
npm ERR!     <http://github.com/npm/npm/issues>

npm ERR! System Linux 3.8.13-bone70
npm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "install" "octalbonescript@1.0.6"
npm ERR! cwd /var/lib/cloud9/grow-io/realgrow
npm ERR! node -v v0.10.29
npm ERR! npm -v 1.4.21
npm ERR! path /var/lib/cloud9/grow-io/realgrow/node_modules/octalbonescript/node_modules/serialport/node_modules/optimist/node_modules/minimist/index.js
npm ERR! fstream_path /var/lib/cloud9/grow-io/realgrow/node_modules/octalbonescript/node_modules/serialport/node_modules/optimist/node_modules/minimist/index.js
npm ERR! fstream_type File
npm ERR! fstream_class FileWriter
npm ERR! code ENOENT
npm ERR! errno 34
npm ERR! fstream_stack /usr/lib/nodejs/fstream/lib/writer.js:284:26
npm ERR! fstream_stack Object.oncomplete (evalmachine.<anonymous>:107:15)
npm ERR! 
npm ERR! Additional logging details can be found in:
npm ERR!     /var/lib/cloud9/grow-io/realgrow/npm-debug.log
npm ERR! not ok code 0

kory44, 2016-03-07:
OK, success. I'm new to linux and didn't realizes you had to worry about versions of software changing so much. I install Debian 7.9 back in December from the beaglebone site, but apparently it was already out of date Here is the site that showed me 3 different ways of fixing this.

https://github.com/fivdi/onoff/wiki/Node.js-v0.10.29-and-native-addons-on-the-BeagleBone-Black

 The problem was that I had node.jsv0.10.29, which is old.

Solution:

run these two obvious lines before attempting to install octalbonescript:
`sudo apt-get update`
`sudo apt-get upgrade`

I'm root so i didn't need the sudo...

closed 2016-03-07 by kory44