theoctal/octalbonescript #59: Doesn't install on mac

whmountains, 2016-03-18:
I can't get the latest version of OBS to install on my mac: 

~~~
npm install --save octalbonescript

> epoll@0.1.17 install /Users/mmc/code/react-dial/node_modules/epoll
> node-gyp rebuild

xcode-select: error: tool 'xcodebuild' requires Xcode, but active developer directory '/Library/Developer/CommandLineTools' is a command line tools instance

xcode-select: error: tool 'xcodebuild' requires Xcode, but active developer directory '/Library/Developer/CommandLineTools' is a command line tools instance

  CXX(target) Release/obj.target/epoll/src/epoll.o
  SOLINK_MODULE(target) Release/epoll.node

> i2c@0.1.8 install /Users/mmc/code/react-dial/node_modules/i2c
> node-gyp rebuild

xcode-select: error: tool 'xcodebuild' requires Xcode, but active developer directory '/Library/Developer/CommandLineTools' is a command line tools instance

xcode-select: error: tool 'xcodebuild' requires Xcode, but active developer directory '/Library/Developer/CommandLineTools' is a command line tools instance

  CXX(target) Release/obj.target/i2c/src/i2c.o
In file included from ../src/i2c.cc:11:
../src/i2c-dev.h:27:10: fatal error: 'linux/types.h' file not found
#include <linux/types.h>
         ^
1 error generated.
make: *** [Release/obj.target/i2c/src/i2c.o] Error 1
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:100:13)
gyp ERR! stack     at ChildProcess.emit (events.js:185:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:200:12)
gyp ERR! System Darwin 15.3.0
gyp ERR! command "/usr/local/Cellar/node/5.6.0/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /Users/mmc/code/react-dial/node_modules/i2c
gyp ERR! node -v v5.6.0
gyp ERR! node-gyp -v v3.3.0
gyp ERR! not ok 
npm WARN install:i2c@0.1.8 i2c@0.1.8 install: `node-gyp rebuild`
npm WARN install:i2c@0.1.8 Exit status 1

> serialport@1.7.4 install /Users/mmc/code/react-dial/node_modules/serialport
> node-pre-gyp install --fallback-to-build

xcode-select: error: tool 'xcodebuild' requires Xcode, but active developer directory '/Library/Developer/CommandLineTools' is a command line tools instance

xcode-select: error: tool 'xcodebuild' requires Xcode, but active developer directory '/Library/Developer/CommandLineTools' is a command line tools instance

  CXX(target) Release/obj.target/serialport/src/serialport.o
In file included from ../src/serialport.cpp:3:
In file included from ../src/serialport.h:5:
../node_modules/nan/nan.h:261:25: error: redefinition of '_NanEnsureLocal'
NAN_INLINE v8::Local<T> _NanEnsureLocal(v8::Local<T> val) {
                        ^
../node_modules/nan/nan.h:256:25: note: previous definition is here
NAN_INLINE v8::Local<T> _NanEnsureLocal(v8::Handle<T> val) {
                        ^
../node_modules/nan/nan.h:661:13: error: no member named 'smalloc' in namespace 'node'
    , node::smalloc::FreeCallback callback
      ~~~~~~^
../node_modules/nan/nan.h:672:12: error: no matching function for call to 'New'
    return node::Buffer::New(v8::Isolate::GetCurrent(), data, size);
           ^~~~~~~~~~~~~~~~~
/Users/mmc/.node-gyp/5.6.0/include/node/node_buffer.h:31:40: note: candidate function not viable: no known
      conversion from 'uint32_t' (aka 'unsigned int') to 'enum encoding' for 3rd argument
NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate,
                                       ^
[... 146 lines cut ...]

npm ERR! Darwin 15.3.0
npm ERR! argv "/usr/local/Cellar/node/5.6.0/bin/node" "/usr/local/bin/npm" "install" "--save" "octalbonescript"
npm ERR! node v5.6.0
npm ERR! npm  v3.8.0
npm ERR! code ELIFECYCLE

npm ERR! serialport@1.7.4 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the serialport@1.7.4 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs serialport
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /Users/mmc/code/react-dial/npm-debug.log
~~~

whmountains, 2016-03-18:
I'm on mac OS 10.11.3

adityapatadia, 2016-03-22:
Updating after checking in a few hours. You will be able to install after that.

closed 2016-03-22 by adityapatadia