theoctal/octalbonescript #61: Can't load SPI cape

RudyJM, 2016-03-22:
Can't load ADAFRUIT-SPI0 using b.loadCape('ADAFRUIT-SPI0');

This is all I have in my node js file (test.js)
var b = require('octalbonescript');

console.log('unload hdmi cape');
b.unloadCape('cape-univ-hdmi');

console.log('load spi cape')
b.loadCape('ADAFRUIT-SPI0');

console.log('done!')

Running the latest version of Debian Jessie for the BBB

root@beaglebone:/# DEBUG=bone node test.js
  bone is_ocp() = /sys/devices/platform/ocp +0ms
  bone find_sysfsFile(cape-universal,/sys/devices/platform/ocp,ocp:cape-universal) +94ms
  bone is_cape_universal() = /sys/devices/platform/ocp/ocp:cape-universal +41ms
  bone Loading HDMI Cape... +10ms
  bone load_dt_sync(cape-univ-hdmi) +15ms
  bone onFindCapeMgr: path = /sys/devices/platform/bone_capemgr +6ms
  bone onReadSlots: index = -1, readAttempts = 1 +17ms
  bone Writing cape-univ-hdmi to /sys/devices/platform/bone_capemgr/slots +5ms
  bone onReadSlots: index = 252, readAttempts = 2 +6s
  bone cape-univ-hdmi is successfully loaded +7ms
  bone load_dt_sync return +6ms
  bone Using Universal Cape interface +11ms
  bone Enabling analog inputs +167ms
  bone load_dt_sync(BB-ADC) +15ms
  bone onFindCapeMgr: path = /sys/devices/platform/bone_capemgr +5ms
  bone onReadSlots: index = 190, readAttempts = 1 +4ms
  bone BB-ADC is successfully loaded +6ms
  bone load_dt_sync return +2ms
  bone find_sysfsFile(helper,/sys/devices/platform/ocp,44e0d000.tscadc) +8ms
  bone analog enable. Path = /sys/devices/platform/ocp/44e0d000.tscadc/TI-am335x-adc/iio:device0/in_voltage +15ms
  bone index.js loaded +8ms
unload hdmi cape
  bone unload_dt_sync(cape-univ-hdmi) +98ms
  bone Found slot : 15 +12ms
  bone Attempting to unload conflicting slot 15 for cape-univ-hdmi +5ms
  bone slot unload successful +121ms
load spi cape
  bone load_dt_sync(ADAFRUIT-SPI0) +11ms
  bone onFindCapeMgr: path = /sys/devices/platform/bone_capemgr +8ms
  bone onReadSlots: index = -1, readAttempts = 1 +9ms
  bone Writing ADAFRUIT-SPI0 to /sys/devices/platform/bone_capemgr/slots +10ms

/usr/local/lib/node_modules/octalbonescript/lib/bone.js:205
                throw new verror(err2, 'Write to CapeMgr slots failed');
                      ^
VError: Write to CapeMgr slots failed: EINVAL, invalid argument
    at onWriteSlots (/usr/local/lib/node_modules/octalbonescript/lib/bone.js:205:23)
    at onReadSlots (/usr/local/lib/node_modules/octalbonescript/lib/bone.js:196:17)
    at readSlots (/usr/local/lib/node_modules/octalbonescript/lib/bone.js:176:13)
    at Object.module.exports.load_dt_sync (/usr/local/lib/node_modules/octalbonescript/lib/bone.js:167:9)
    at Object.f.loadCape (/usr/local/lib/node_modules/octalbonescript/index.js:56:15)
    at Object.<anonymous> (/test.js:7:3)
    at Module._compile (module.js:456:26)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
root@beaglebone:/# 

root@beaglebone:/# cat /sys/devices/platform/bone_capemgr/slots
 0: PF----  -1 
 1: PF----  -1 
 2: PF----  -1 
 3: PF----  -1 
 4: P-O-L-   0 Override Board Name,00A0,Override Manuf,cape-universaln
 6: P-O-L-   2 Override Board Name,00A0,Override Manuf,BB-ADC
root@beaglebone:/# 

Adafruit dts
https://github.com/adafruit/adafruit-beaglebone-io-python/blob/master/overlays/ADAFRUIT-SPI0-00A0.dts

adityapatadia, 2016-03-22:
I will have to check but I am busy this week. I will take a look next week.

RudyJM, 2016-03-22:
Great, thank you!

closed 2016-03-23 by adityapatadia