theoctal/octalbonescript #63: OBS doesn't work with LCD Cape

nathanpjones, 2016-03-30:
I can't manage to get OBS working with a LCD cape ([4DCAPE-70T](http://www.4dsystems.com.au/product/4DCAPE_70T/)) on kernel 4.1. Because OBS loads the universal cape at startup, I get the following error.

```
  bone is_ocp() = /sys/devices/platform/ocp +0ms
  bone find_sysfsFile(cape-universal,/sys/devices/platform/ocp,ocp:cape-universal) +24ms
  bone is_cape_universal() = null +11ms
  bone Loading Universal Cape interface... +2ms
  bone load_dt_sync(cape-universaln) +6ms
  bone onFindCapeMgr: path = /sys/devices/platform/bone_capemgr +2ms
  bone onReadSlots: index = -1, readAttempts = 1 +6ms
  bone Writing cape-universaln to /sys/devices/platform/bone_capemgr/slots +2ms

/var/lib/cloud9/projects/led_test/node_modules/octalbonescript/lib/bone.js:205
                throw new verror(err2, 'Write to CapeMgr slots failed');
                      ^
VError: Write to CapeMgr slots failed: EEXIST, file already exists
    at onWriteSlots (/var/lib/cloud9/projects/led_test/node_modules/octalbonescript/lib/bone.js:205:23)
    at onReadSlots (/var/lib/cloud9/projects/led_test/node_modules/octalbonescript/lib/bone.js:196:17)
    at readSlots (/var/lib/cloud9/projects/led_test/node_modules/octalbonescript/lib/bone.js:176:13)
    at Object.module.exports.load_dt_sync (/var/lib/cloud9/projects/led_test/node_modules/octalbonescript/lib/bone.js:167:9)
    at Object.<anonymous> (/var/lib/cloud9/projects/led_test/node_modules/octalbonescript/index.js:31:12)
    at Module._compile (module.js:456:26)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
    at Module.require (module.js:364:17)
```

These are the current capes.

```
 0: P---L-   0 4D 7.0 LCD CAPE- 4DCAPE-70T     ,00A3,4D SYSTEMS      ,BB-BONE-LCD7-01
 1: PF----  -1
 2: PF----  -1
 3: PF----  -1
```

My program works if I disable the cape in `uEnv.txt` or if I physically unplug it. When disabling I've also tried loading the cape after successful init, but it has no effect and obviously having it disabled at startup would affect my ability to boot to a GUI.

Is there some way to start up without loading the universal cape?

adityapatadia, 2016-04-02:
There is a way. `AUTO_LOAD_CAPE=0 node myscript.js`

closed 2016-04-02 by adityapatadia

nathanpjones, 2016-04-06:
Thanks for the response and for updating the wiki. However, it's still not clear how OBS provides for this situation.

So now I'm able to forgo loading the universal cape and load the cape for my LCD screen. However, when I go to use one of the GPIO pins not associated with the cape (`P8_08`) I get an error that the pin is not available to write.

I've tried loading the universal cape after my LCD's cape, but I end up in the same situation as before. I poked around @psiphi75's repos but he apparently hasn't gotten around to using `AUTO_LOAD_CAPE` yet.

Is there some way to work with OBS without the cape paradigm or do I have to create a custom cape definition for my scenario? Or is there something I'm missing?

psiphi75, 2016-04-07:
Have you tried loading a cape that registers the `P8_08` pin for your purpose?  If you don't load a cape that registers that pin it will not be accessible.  I have had a similar problem, in-fact that is why I created the `AUTO_LOAD_CAPE` solution.

psiphi75, 2016-04-07:
@adityapatadia, it appears 1.1.2 version of the OBS package on npmjs.org does not have the most recent changes.  Can you please republish this.
@nathanpjones, the [bbg-rc.local](https://github.com/psiphi75/studious-octo-guide/blob/master/the-whole-shebang/boot/bbg-rc.local) file has an example of `AUTO_LOAD_CAPE` in action.

nathanpjones, 2016-04-07:
@psiphi75 Thanks for the reply. I can use `P8_08` by unplugging the LCD and loading the universal cape at both the system and OBS level. From what you're saying, it sounds like I have to create a cape definition that draws from the universal and excludes pins already loaded by my LCD cape.

Also, looking at your use case I see that even though you prevent the universal cape from being loaded automatically you just still end up loading it directly. What does that accomplish?

psiphi75, 2016-04-07:
@nathanpjones There may already be an overlay in `/lib/firmware` which initialises `P8_08`, .  There is a [BBB Device Tree Overlay Generator](http://kilobaser.com/blog/2014-07-28-beaglebone-black-devicetreeoverlay-generator), I have not used it, but it looks promising.  [I see](http://au.mathworks.com/help/supportpkg/beagleboneio/ug/beaglebone-black-pin-map.html) that `P8_08` has two functions, as GPIO and TIMER7.
The reason I use `AUTO_LOAD_CAPE` is because I only require the universal cape, not the others.  It takes time to load capes and I want my device to boot as quickly as possible.

adityapatadia, 2016-04-08:
I will publish latest version today.

adityapatadia, 2016-04-08:
Published.