theoctal/octalbonescript #65: Debian 8.3: VError: Write to CapeMgr slots failed: EEXIST, file already exists

Shadocko, 2016-04-25:
Hello,

I'm trying to use some of the HDMI pins in GPIO mode with an up-to-date debian 8.3 but have been unable to so far.
In /boot/uEnv.txt, I have the following settings:

    ##BeagleBone Black: HDMI (Audio/Video) disabled:
    dtb=am335x-boneblack-emmc-overlay.dtb

    cmdline=coherent_pool=1M quiet cape_universal=enable

After booting, I get the following cape overlay:

    cat /sys/devices/platform/bone_capemgr/slots
     0: PF----  -1
     1: PF----  -1
     2: PF----  -1
     3: PF----  -1
     4: P-O-L-   0 Override Board Name,00A0,Override Manuf,univ-emmc

The pins can seemingly be interacted with using config-pin:

    root@beaglebone:~# config-pin -l P9.29
    default gpio gpio_pu gpio_pd pwm spi pruout pruin
    root@beaglebone:~# config-pin P9.29 pwm
    root@beaglebone:~#

However, I'm unable to load octalbonescript in this setup:

    /usr/local/lib/node_modules/octalbonescript/lib/bone.js:205
                    throw new verror(err2, 'Write to CapeMgr slots failed');
                          ^
    VError: Write to CapeMgr slots failed: EEXIST, file already exists
        at onWriteSlots (/usr/local/lib/node_modules/octalbonescript/lib/bone.js:205:23)
        at onReadSlots (/usr/local/lib/node_modules/octalbonescript/lib/bone.js:196:17)
        at readSlots (/usr/local/lib/node_modules/octalbonescript/lib/bone.js:176:13)
        at Object.module.exports.load_dt_sync (/usr/local/lib/node_modules/octalbonescript/lib/bone.js:167:9)
        at Object.<anonymous> (/usr/local/lib/node_modules/octalbonescript/index.js:47:18)
        at Module._compile (module.js:456:26)
        at Object.Module._extensions..js (module.js:474:10)
        at Module.load (module.js:356:32)
        at Function.Module._load (module.js:312:12)
        at Module.require (module.js:364:17)

If I disable the univ-emmc dto and instead use the default cape-univeraln one, octalbonescript will cause cape-hdmi to be loaded and will error on my attempt to use P9_29 as PWM.
How can the HDMI pins be used with octalbonescript?

Andrewiski, 2016-05-17:
try setting AUTO_LOAD_CAPE  so octalbone doesn't try to autoload the cape

 `sudo DEBUG=bone AUTO_LOAD_CAPE=0 node app.js`

fafamendes92, 2016-05-31:
Hi Andrewiski,

Thanks for the solution!
Would have some way to definitely solve?

closed 2016-05-31 by adityapatadia