theoctal/octalbonescript #67: Write to CapeMgr slots failed: EINVAL, invalid argument

Andrewiski, 2016-05-13:
steps to reproduce.  

Fresh lxqt-4gb Linux beaglebone 4.4.8-ti-r22 #1 SMP Wed Apr 27 22:23:10 UTC 2016

```
apt-get update
apt-get upgrade

systemctl disable cloud9.service
systemctl disable gateone.service
systemctl disable bonescript.service
systemctl disable bonescript.socket
systemctl disable bonescript-autorun.service
systemctl disable avahi-daemon.service
systemctl disable gdm.service
systemctl disable mpd.service
```

app.js
```
var boneScript;
   
try {
    boneScript = require('octalbonescript');
    boneScript.getPlatform(function (err, x) {
        console.log('bonescript getPlatform');
        console.log('version = ' + x.version);
        console.log('serialNumber = ' + x.serialNumber);
        console.log('dogtag = ' + x.dogtag);
    });
} catch (e) {
    console.log(e);
}
```

package.js

```
{
  "name": "test-bonescript",
  "version": "0.0.0",
    "description": "testBonescript",
    "scripts": {
    "start": "node ./app.js",
    "stop": "pkill --signal SIGINT testBoneScript"
  },
  "main": "app.js",
    "author": {
        "name": "adevries",
        "email": "adevries at digitalexample.com"
    },
    "dependencies": {
        "octalbonescript": "^1.2.0"
    }
}
```

```
npm install
sudo DEBUG=bone node app.js
```

 bone Using Universal Cape interface +0ms
  bone is_ocp() = /sys/devices/platform/ocp +83ms
  bone find_sysfsFile(cape-universal,/sys/devices/platform/ocp,ocp:cape-universal) +20ms
  bone is_cape_universal() = null +8ms
  bone Loading Universal Cape interface... +2ms
  bone load_dt_sync(cape-universaln) +6ms
  bone onFindCapeMgr: path = /sys/devices/platform/bone_capemgr +3ms
  bone onReadSlots: index = -1, readAttempts = 1 +7ms
  bone Writing cape-universaln to /sys/devices/platform/bone_capemgr/slots +2ms
{ [VError: Write to CapeMgr slots failed: EINVAL, invalid argument]
  jse_shortmsg: 'Write to CapeMgr slots failed',
  jse_summary: 'Write to CapeMgr slots failed: EINVAL, invalid argument',
  jse_cause: { [Error: EINVAL, invalid argument] errno: -22, code: 'EINVAL', syscall: 'write' },
  message: 'Write to CapeMgr slots failed: EINVAL, invalid argument' }

Andrewiski, 2016-05-17:
a check of ‘cat /sys/devices/platform/bone_capemgr/slots’  output is
0: PF----  -1
 1: PF----  -1
 2: PF----  -1
 3: PF----  -1

Not how no universal cape is loaded.

Console Debug from unchanged image 2016-05-01  shows error 
Could not find symbol 'pruss'
[   25.116787] bone_capemgr bone_capemgr: slot #4: Failed to resolve tree

Following the instructions on https://github.com/RobertCNelson/bb.org-overlays

I did a sudo /opt/scripts/tools/update_kernel.sh --lts-4_4 --bone-channel

now the 'pruss' error above is resolved and ‘cat /sys/devices/platform/bone_capemgr/slots’  output is

0: PF----  -1
 1: PF----  -1
 2: PF----  -1
 3: PF----  -1
 4: P-O-L-   0 Override Board Name,00A0,Override Manuf,univ-emmc

a retry of above results are 

debian@beaglebone:/var/testBonescript$ sudo DEBUG=bone node app.js
trying OctalboneScript
  bone Using Universal Cape interface +0ms
  bone is_ocp() = /sys/devices/platform/ocp +85ms
  bone find_sysfsFile(cape-universal,/sys/devices/platform/ocp,ocp:cape-universal) +18ms
  bone is_cape_universal() = /sys/devices/platform/ocp/ocp:cape-universal +7ms
  bone Loading HDMI Cape... +6ms
  bone load_dt_sync(cape-univ-hdmi) +4ms
  bone onFindCapeMgr: path = /sys/devices/platform/bone_capemgr +1ms
  bone onReadSlots: index = -1, readAttempts = 1 +5ms
   bone_capemgr bone_capemgr: slot #5: cape-univ-hdmi conflict P8.27 (#4:univ-emmc)
[32;1mbone Writing cape-univ-hdmi to /sys/devices/platform/b[ 5901.954752] bone_capemgr bone_capemgr: slot #5: Failed verification
one_capemgr/slots +1ms
{ [VError: Write to CapeMgr slots failed: EEXIST, file already exists]
  jse_shortmsg: 'Write to CapeMgr slots failed',
  jse_summary: 'Write to CapeMgr slots failed: EEXIST, file already exists',
  jse_cause: { [Error: EEXIST, file already exists] errno: -17, code: 'EEXIST', syscall: 'write' },
  message: 'Write to CapeMgr slots failed: EEXIST, file already exists' }

So progress on that front but the next error is because it is trying to force the load of unversial when it is already loaded.  

per @RobertNelson  we should check to see if unversial is already loaded instead of just loading it

https://github.com/RobertCNelson/boot-scripts/blob/master/boot/am335x_evm.sh#L321-L407

I will attempt a patch for this logic

Andrewiski, 2016-05-17:
env check is already there to prevent autoload of cape and based on what I see with base Debian Image whcih already auto loads the universial Cape.  I would think that should be the default behavior.

Anyways setting AUTO_LOAD_CAPE=0  disables the load of the already loaded cape and no more errors

`sudo DEBUG=bone AUTO_LOAD_CAPE=0 node app.js`

Andrewiski, 2016-05-17:
Newer Debian images got posted today with updated Kernels so default image now works as expected but AUTO_LOAD_CAPE=0 is still needed to prevent the attempt to load an already loaded cape on the Beaglebone Green

adityapatadia, 2016-05-18:
Let me get back to you this Saturday. I am sorry it's hard to reply and resolve issue during weekdays.

Andrewiski, 2016-06-09:
Pull request #69 posted with both patches for i2c and universal cape autoload fixes.
@adityapatadia

closed 2016-08-19 by adityapatadia