theoctal/octalbonescript #68: Multiple i2c Devices same device

Andrewiski, 2016-05-31:
I2c allows multiple same on same linux device with different i2c addresses.  Because the port exposed via the i2c.open callback does not expose .setAddress and you are indexing the opened i2c by device "'/dev/i2c-1" and not '/dev/i2c-1' + address  calls to open multiple i2c slaves with different address causes all previous devices to have the address of the last device opened.  With both of these limitations I do not see how it is possible to open two i2c slaves with different addresses on the same linux i2c device.

You should be indexing on both path and address.

ie

`i2c.openPorts[path + '/' + address] = port;`

`open: function(path, address, handler, callback) {
		debug("i2c.open(" + path + ", " + address + ")");

		if (!i2c.ports[path]) {
			throw new verror("Supplied path:" + path + " is not a vaild i2c port path.");
		}

		if (typeof handler != "function" || typeof callback != 'function') {
			throw new verror("handler and callback must be provided to i2c.open and both should be functions");
		}

		module.exports.enable(path, onI2CEnable);

		function onI2CEnable(err) {
			if (err) {
				throw new verror(err);
			}

			if (!i2c.openPorts[path]) {
				var port = new i2cPort(address, {
					device: i2c.ports[path].path
				});
				debug('opened i2c port: ' + path);
				port.on('data', handler);
				i2c.openPorts[path] = port;
				if (callback) callback(null, i2c.openPorts[path]);
			} else {
				if (callback) callback(null, i2c.openPorts[path]);
			}
		}
	},`

adityapatadia, 2016-05-31:
Can you please create PR if you are sure how to solve this issue?

Andrewiski, 2016-05-31:
Sure I can make a Pull Request for fixing multiple I2c on same device.
Would you like me to have a go at fixing the loading of universial capes using  
@RobertNelson logic from his boot script as well

https://github.com/RobertCNelson/boot-scripts/blob/master/boot/am335x_evm.sh#L321-L407

RobertNelson, 2016-05-31:
Hey guys,

@RobertNelson is NOT @RobertCNelson. :-)

While I have the deepest respect for boot scripts, I can’t be of much help writing or modifying them.

Thanks,

@RobertNelson

adityapatadia, 2016-06-01:
Hi @Andrewiski what do you mean by fixing universal cape loading? You can work and create PR if you feel it will improve OBS. I almost always merge PRs which are crafted well.

Andrewiski, 2016-06-02:
Pull request https://github.com/theoctal/octalbonescript/pull/69  posted.  Not breaking change on some methods as address must be passed in with path but suspect most users of the open function so not call your methods anyways just use the returned port objects.

Andrewiski, 2016-06-02:
Pull request https://github.com/theoctal/octalbonescript/pull/69  posted.  Note breaking change on some methods as address must be passed in with path but suspect most users of the open function so not call your methods anyways just use the returned port objects.

Andrewiski, 2016-06-13:
@adityapatadia   Did you get a chance to look over my suggested changes to fix multiple calls to i2c and fix for Universal Cape already being loaded?

adityapatadia, 2016-06-14:
I am sorry I could not get time. I will give a shot this weekend.

Andrewiski, 2016-07-03:
@adityapatadia  Any update on this pull request or should I just start using my own branch in my applications.

closed 2016-08-19 by adityapatadia