theoctal/octalbonescript #71: Serial Port Issues when running OctalBoneScript on newest debian

murban1, 2016-10-20:
Good day,

I've recently been trying to run some code that I have successfully run on a previously setup beaglebone. I flashed a fresh beaglebone with the debian 8.5 from [Here](https://debian.beagleboard.org/images/bone-debian-8.4-lxqt-4gb-armhf-2016-05-13-4gb.img.xzl).

After successfully installing, I updated and upgraded and ended up with the following:

`root@beaglebone:~/beaglebone# uname -a`
`Linux beaglebone 4.4.12-ti-r31 #1 SMP Thu Jun 16 18:48:27 UTC 2016 armv7l GNU/Linux`

`npm -v`
`3.10.8`

`node -v`
`v0.12.16`

Everything looks good, but after installing octalbonescript with no errors, I try and run my code:
`DEBUG=bone node beaglebone.js `
`Optional package 'serialport' not loaded
/usr/local/lib/node_modules/octalbonescript/lib/serial.js:98
        parsers: serialPort.exists ? serialPort.parsers : {}
                           ^
TypeError: Cannot read property 'exists' of undefined
    at Object.<anonymous> (/usr/local/lib/node_modules/octalbonescript/lib/serial.js:98:21)
    at Module._compile (module.js:460:26)
    at Object.Module._extensions..js (module.js:478:10)
    at Module.load (module.js:355:32)
    at Function.Module._load (module.js:310:12)
    at Module.require (module.js:365:17)
    at require (module.js:384:17)
    at Object.<anonymous> (/usr/local/lib/node_modules/octalbonescript/index.js:11:14)
    at Module._compile (module.js:460:26)
    at Object.Module._extensions..js (module.js:478:10)
`

The last time I tried to flash and install, it was complaining about the I2C module in the same manner.

I'm not sure what would be causing this issue, so any help would be appreciated.

adityapatadia, 2016-10-28:
It seems serialport module is not loaded. It seems they have different versions for different platforms and kernels. Can you identify which one will suite OBS? https://www.npmjs.com/package/serialport

dceejay, 2017-01-25:
the latest serialport 4.x compiles OK on node.js 6.9.x LTS... any chance of moving to that ?

adityapatadia, 2017-01-26:
We can move to that but need to test the code once integrated. I currently don't have time for that. May be I will do after 10-15 days. Can you create PR?

troywweber7, 2017-01-27:
Any progress on this? I have difficulty with `npm i -g octalbonescript` as well.

```
root@beaglebone:~# uname -a
Linux beaglebone 4.4.30-ti-r64 #1 SMP Fri Nov 4 21:23:33 UTC 2016 armv7l GNU/Linux
root@beaglebone:~# node -v
v0.12.18
root@beaglebone:~# npm -v
2.15.11
```

It tries to install but ends up with the following error:

```
Failed to execute '/usr/bin/nodejs /usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/usr/local/lib/node_modules/octalbonescript/node_modules/serialport/build/Release/node-v14-linux-arm/serialport.node --module_name=serialport --module_path=/usr/local/lib/node_modules/octalbonescript/node_modules/serialport/build/Release/node-v14-linux-arm' (1)
npm ERR! Linux 4.4.30-ti-r64
npm ERR! argv "/usr/bin/node" "/usr/bin/npm" "i" "-g" "octalbonescript"
npm ERR! node v0.12.18
npm ERR! npm  v2.15.11
npm ERR! code ELIFECYCLE

npm ERR! serialport@2.0.6 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the serialport@2.0.6 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! This is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs serialport
npm ERR! Or if that isn't available, you can get their info via:
npm ERR! 
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /var/lib/cloud9/npm-debug.log
```

adityapatadia, 2017-01-30:
@dceejay if I move code to newer serialport, can you check it on BBB? I don't have board around to check right now.

dceejay, 2017-02-01:
sure

adityapatadia, 2017-02-02:
I updated code in master branch. Can you check if it works on real BeagleBone?

dceejay, 2017-02-02:
So far seems fine on both node 4.7.2 and node 6.9.5
(does have some deprecation warnings on 6.9.5 - but that is their problem not yours)

adityapatadia, 2017-02-03:
Ok, I am releasing new version and closing this issue.

closed 2017-02-03 by adityapatadia

dceejay, 2017-02-04:
Thanks very much.

ArashBJ, 2017-02-10:
Hi to everyone.
I recently update my OBS to 1.3.0 and had the same issue as the one in the topic.
When i load OBS by require command in my JS file, program crashes on line 98 of lib/serial.js

And i don't think the problem is because of serial module not loaded correctly!
@adityapatadia 

I checked changes on the serial.js file and found following changes on line 7:
```
-     var serialPort = bone.require('serialport').SerialPort;
+    var SerialPort = bone.require('serialport');
```
And after that line 69 was corrected in following form:
```
-				var port = new serialPort(path, options);
+				var port = new SerialPort(path, options);
```
But line 98 still in old manner:
`parsers: serialPort.exists ? serialPort.parsers : {}`

So i changed it as following and everything is OK now:
`parsers: SerialPort.exists ? SerialPort.parsers : {}`

My Beaglebone Black software and modules versions:
```
root@beaglebone:/# uname -a
Linux beaglebone 4.4.36-ti-r72 #1 SMP Wed Dec 7 22:29:53 UTC 2016 armv7l GNU/Linux

root@beaglebone:/# node -v
v4.7.0

root@beaglebone:/# npm -v
2.15.11

root@beaglebone:/# npm list -g octalbonescript
/usr/local/lib
└── octalbonescript@1.3.0 

root@beaglebone:/# cd /usr/local/lib/node_modules/octalbonescript
root@beaglebone:/usr/local/lib/node_modules/octalbonescript# npm list serialport
octalbonescript@1.3.0 /usr/local/lib/node_modules/octalbonescript
└── serialport@4.0.7 
```

adityapatadia, 2017-02-10:
Can you create PR?

Regards,
Aditya Patadia

adityapatadia, 2017-02-16:
I pushed v1.3.1 fixing this issue.