theoctal/octalbonescript #72: SPI Support in the Octal Bone Script Library?

totem415, 2017-05-25:
Any thoughts on how my project should do SPI communication?  I'm worried that 3rd party node modules would conflict since they are accessing pins to communicate.  Any help or suggestions would be greatly appreciated.

BTW the OBS is working great for our needs so far.  Thanks so much for developing and maintaining!

-roger

adityapatadia, 2017-06-07:
I don't think SPI is currently possible with code as is. You should find a device tree and load it to work on it. https://github.com/beagleboard/bb.org-overlays/tree/master/src/arm

I am afraid lot of code changes needs to be done to support this and I currently don't have time to work on it. If you are willing to do, please create PR.

adityapatadia, 2017-12-24:
This library is abandoned. Please check notice in README for further information.

closed 2017-12-24 by adityapatadia