theoctal/octalbonescript #73: developing off-board with simulator

troywweber7, 2017-06-02:
Octalbonescript claims to have a simulation mode (although it appears to only work on linux and not Mac or Windows). I thought maybe this would make it easy to develop for the beaglebone off-board but I cannot find documentation ANYWHERE on how to enable the simulator.

Will I have to write my own object that "mocks" the functions provided by octalbonescript for when I am developing off the beaglebone?

adityapatadia, 2017-06-07:
You run it on Mac or Linux on unsupported hardware (i.e. Macbook or Laptop) and it will automatically switch to simulator mode.

troywweber7, 2017-06-14:
@adityapatadia What does switching to simulator mode entail then? When I try to  run an octalbonescript command on Linux, it throws errors which leads me to believe the simulator is not working.

troywweber7, 2017-06-20:
@adityapatadia it throws this error: `The pin P8_03 is not availble to write. Please make sure it is not used by another cape.`

adityapatadia, 2017-12-24:
This library is abandoned. Please check notice in README for further information.

closed 2017-12-24 by adityapatadia