theoctal/octalbonescript #74: VError: Write to CapeMgr slots failed: EACCES: permission denied, open '/sys/devices/platform/bone_capemgr/slots'

troywweber7, 2017-06-14:
I've freshly installed the official BeagleBone Green Wireless image on my BBGW, so I'm running Debian 8.6.

I've fully updated the software on the board multiple times to make sure I am up to date with `sudo apt-get update && sudo apt-get -y upgrade`.

Also, I created a new user `deploy` and have sudo permissions. As `deploy` I've installed Node 7.10 using `nvm`.

In a test folder, I initialized a simple `package.json` and `npm i -S octalbonescript`. It installs fine, so I go into the node interpreter and try to import octalbonescript and havoc ensues:

```bash
$ cat package.json 
{
  "name": "test-octalbonescript",
  "version": "1.0.0",
  "description": "",
  "main": "index.js",
  "scripts": {
    "test": "echo \"Error: no test specified\" && exit 1"
  },
  "author": "",
  "license": "ISC",
  "dependencies": {
    "octalbonescript": "^1.3.1"
  }
}
$ clear

$ node
> obs = require('octalbonescript')
VError: Write to CapeMgr slots failed: EACCES: permission denied, open '/sys/devices/platform/bone_capemgr/slots'
  at onWriteSlots (/home/deploy/code/test-octalbonescript/node_modules/octalbonescript/lib/bone.js:289:23)
  at onReadSlots (/home/deploy/code/test-octalbonescript/node_modules/octalbonescript/lib/bone.js:280:17)
  at readSlots (/home/deploy/code/test-octalbonescript/node_modules/octalbonescript/lib/bone.js:260:13)
  at Object.load_dt_sync (/home/deploy/code/test-octalbonescript/node_modules/octalbonescript/lib/bone.js:251:9)
  at Object.load_universal_cape (/home/deploy/code/test-octalbonescript/node_modules/octalbonescript/lib/bone.js:144:24)
  at Object.<anonymous> (/home/deploy/code/test-octalbonescript/node_modules/octalbonescript/index.js:37:18)
  at Module._compile (module.js:571:32)
  at Object.Module._extensions..js (module.js:580:10)
  at Module.load (module.js:488:32)
  at tryModuleLoad (module.js:447:12)
  at Function.Module._load (module.js:439:3)
  at Module.require (module.js:498:17)
  at require (internal/module.js:20:19)
  at repl:1:7
  at ContextifyScript.Script.runInThisContext (vm.js:23:33)
  at REPLServer.defaultEval (repl.js:339:29)
  at bound (domain.js:280:14)
  at REPLServer.runBound [as eval] (domain.js:293:12)
  at REPLServer.onLine (repl.js:536:10)
  at emitOne (events.js:101:20)
  at REPLServer.emit (events.js:191:7)
  at REPLServer.Interface._onLine (readline.js:241:10)
  at REPLServer.Interface._line (readline.js:590:8)
  at REPLServer.Interface._ttyWrite (readline.js:869:14)
  at REPLServer.self._ttyWrite (repl.js:609:7)
  at ReadStream.onkeypress (readline.js:120:10)
  at emitTwo (events.js:106:13)
  at ReadStream.emit (events.js:194:7)
  at emitKeys (internal/readline.js:402:14)
  at emitKeys.next (<anonymous>:null:null)
  at ReadStream.onData (readline.js:980:36)
  at emitOne (events.js:96:13)
  at ReadStream.emit (events.js:191:7)
  at readableAddChunk (_stream_readable.js:178:18)
  at ReadStream.Readable.push (_stream_readable.js:136:10)
  at TTY.onread (net.js:561:20)

> 
(To exit, press ^C again or type .exit)
> 
```

troywweber7, 2017-06-14:
So if I set the group of the file giving the EACCESS error to `gpio`, and have myself added to `gpio` group, and change the mod with `chmod g+w` then obs loads within Node 7.10 just fine. I can do the same process for the `USR0` led. But I need to do this to every pin that might be accessed from octalbonescript in Node 7.10. Can anyone point me to documentation on how to achieve this so it is persistent across restarts?

albham, 2017-11-16:
I solved this issue defining the file `/etc/udev/rules.d/75-bone_capemgr-noroot.rules`, 
based in `80-gpio-noroot.rules`,  with content:

```
# Change group to gpio
SUBSYSTEM=="platform", PROGRAM="/bin/sh -c '/bin/chown -R root:gpio /sys/devices/platform/bone_capemgr'"
# Change user permissions to ensure user and group have read/write permissions
SUBSYSTEM=="platform", PROGRAM="/bin/sh -c '/bin/chmod -R ug+rw /sys/devices/platform/bone_capemgr'"
```

adityapatadia, 2017-12-24:
This library is abandoned. Please check notice in README for further information.

closed 2017-12-24 by adityapatadia