theoctal/octalbonescript #75: loading univ-bbgw doesn't allows access to all gpio pins

troywweber7, 2017-06-15:
I am new to cape and cape manager, but I know enough now to load `univ-bbgw` on my beaglebone green wireless. I have a test script that checks which pins I can set to output:

```js
#!/usr/bin/env node

process.env['AUTO_LOAD_CAPE'] = 0;

console.log('loading octalbonescript...');
const obs = require('octalbonescript');

console.log('loading cape...');
obs.loadCape('univ-bbgw');

const fs = require('fs');

// fs.writeFileSync('pinmap.json', JSON.stringify(obs.pinmap));

let gpio_pins = Object.keys(obs.pinmap).filter(k => obs.pinmap[k].gpio >= 0);

for (let pin of gpio_pins)
{
	let message = `${pin}: `;

	try
	{
		obs.pinModeSync(pin, obs.OUTPUT);
		message += `set to ${obs.OUTPUT}`
	}
	catch (e)
	{
		message += e.message || e;
	}

	console.log(message);
}
```

When I run this, the output is:

```
loading octalbonescript...
loading cape...
USR0: set to out
USR1: set to out
USR2: set to out
USR3: set to out
P8_03: The pin P8_03 is not availble to write. Please make sure it is not used by another cape.
P8_04: The pin P8_04 is not availble to write. Please make sure it is not used by another cape.
P8_05: The pin P8_05 is not availble to write. Please make sure it is not used by another cape.
P8_06: The pin P8_06 is not availble to write. Please make sure it is not used by another cape.
P8_07: set to out
P8_08: set to out
P8_09: set to out
P8_10: set to out
P8_11: The pin P8_11 is not availble to write. Please make sure it is not used by another cape.
P8_12: The pin P8_12 is not availble to write. Please make sure it is not used by another cape.
P8_13: set to out
P8_14: The pin P8_14 is not availble to write. Please make sure it is not used by another cape.
P8_15: The pin P8_15 is not availble to write. Please make sure it is not used by another cape.
P8_16: The pin P8_16 is not availble to write. Please make sure it is not used by another cape.
P8_17: The pin P8_17 is not availble to write. Please make sure it is not used by another cape.
P8_18: The pin P8_18 is not availble to write. Please make sure it is not used by another cape.
P8_19: set to out
P8_20: The pin P8_20 is not availble to write. Please make sure it is not used by another cape.
P8_21: The pin P8_21 is not availble to write. Please make sure it is not used by another cape.
P8_22: The pin P8_22 is not availble to write. Please make sure it is not used by another cape.
P8_23: The pin P8_23 is not availble to write. Please make sure it is not used by another cape.
P8_24: The pin P8_24 is not availble to write. Please make sure it is not used by another cape.
P8_25: The pin P8_25 is not availble to write. Please make sure it is not used by another cape.
P8_26: The pin P8_26 is not availble to write. Please make sure it is not used by another cape.
P8_27: set to out
P8_28: set to out
P8_29: set to out
P8_30: set to out
P8_31: set to out
P8_32: set to out
P8_33: set to out
P8_34: set to out
P8_35: set to out
P8_36: set to out
P8_37: set to out
P8_38: set to out
P8_39: set to out
P8_40: set to out
P8_41: set to out
P8_42: set to out
P8_43: set to out
P8_44: set to out
P8_45: set to out
P8_46: set to out
P9_11: set to out
P9_12: The pin P9_12 is not availble to write. Please make sure it is not used by another cape.
P9_13: set to out
P9_14: set to out
P9_15: set to out
P9_16: set to out
P9_17: set to out
P9_18: set to out
P9_19: The pin P9_19 is not availble to write. Please make sure it is not used by another cape.
P9_20: The pin P9_20 is not availble to write. Please make sure it is not used by another cape.
P9_21: set to out
P9_22: set to out
P9_23: set to out
P9_24: set to out
P9_25: set to out
P9_26: set to out
P9_27: set to out
P9_28: The pin P9_28 is not availble to write. Please make sure it is not used by another cape.
P9_29: The pin P9_29 is not availble to write. Please make sure it is not used by another cape.
P9_30: The pin P9_30 is not availble to write. Please make sure it is not used by another cape.
P9_31: The pin P9_31 is not availble to write. Please make sure it is not used by another cape.
P9_41: set to out
P9_42: set to out
```
Is there a way to enable the ability to set those not available to outputs as well? The documentation for the BBGW says there are 65 GPIO pins (maybe 67, based on this list)...

RobertCNelson, 2017-08-16:
@troywweber7 the wireless functionality of the bbgw uses many of the expansion pins.  (this issue was resolved for the bbbw).

Regards,

adityapatadia, 2017-12-24:
This library is abandoned. Please check notice in README for further information.

closed 2017-12-24 by adityapatadia