theoctal/octalbonescript #78: still have limited access and no PWM

troywweber7, 2017-08-07:
So in the past I found I needed to load the right cape-overlay to get access to proper pins. And just the other day, based off a script I wrote, I had access to 42 pins for input/output. However, I still can't seem to get access to the PWM and now that number of accessible pins has dropped from 42 to 25.

I'm getting a lot of "**BeagleBone does not allow ANALOG_OUTPUT for pin: P8_07**" and "**Error reading directory /sys/devices/platform/ocp/48304000.epwmss/48304200.ehrpwm/pwm**".

Here are my slots:

```bash
$ cat $SLOTS
 0: PF----  -1 
 1: PF----  -1 
 2: PF----  -1 
 3: PF----  -1 
 4: P-O-L-   0 Override Board Name,00A0,Override Manuf,cape-universaln
```

Here is the script (`test-octalbonescript-pins.js`) I wrote to test:

```js
#!/usr/bin/env node

const obs = require('octalbonescript');
const fs = require('fs');
const path = require('path');
const pins = Object.keys(obs.pinmap)
	.filter((x) => /^USR\d$|^P[89]_\d\d$/.test(x));

// console.log(pins);

const results = pins
	.map((pin) =>
	{
		const modes = ['OUTPUT', 'INPUT_PULLUP', 'INTPUT_PULLDOWN', 'ANALOG_OUTPUT', 'INPUT']
			.filter((mode) =>
			{
				let success = true;
				try
				{
					success = obs.pinModeSync(pin, obs[mode]);
					success = !!success;
				}
				catch (e)
				{
					success = false;
				}
				console.log(`   ${pin} as ${mode}? ${success ? 'pass' : 'fail'}`);
				return success;
			});
		return { pin, modes };
	})
	.filter((obj) => obj.modes.length > 0);

fs.writeFileSync(path.join(__dirname, 'results_pinMode.json'), JSON.stringify(results));
```
With cape-uninversaln loaded, I run `node test-octalbonescript-pins.js | grep pass`:

```bash
$ node test-octalbonescript-pins.js | grep pass
   USR0 as OUTPUT? pass
pinMode only supports GPIO output for LED pin: USR0
BeagleBone does not allow ANALOG_OUTPUT for pin: USR0
pinMode only supports GPIO output for LED pin: USR0
   USR1 as OUTPUT? pass
pinMode only supports GPIO output for LED pin: USR1
BeagleBone does not allow ANALOG_OUTPUT for pin: USR1
pinMode only supports GPIO output for LED pin: USR1
   USR2 as OUTPUT? pass
pinMode only supports GPIO output for LED pin: USR2
BeagleBone does not allow ANALOG_OUTPUT for pin: USR2
pinMode only supports GPIO output for LED pin: USR2
   USR3 as OUTPUT? pass
pinMode only supports GPIO output for LED pin: USR3
BeagleBone does not allow ANALOG_OUTPUT for pin: USR3
pinMode only supports GPIO output for LED pin: USR3
   P8_07 as OUTPUT? pass
   P8_07 as INPUT_PULLUP? pass
BeagleBone does not allow ANALOG_OUTPUT for pin: P8_07
   P8_07 as INPUT? pass
   P8_08 as OUTPUT? pass
   P8_08 as INPUT_PULLUP? pass
BeagleBone does not allow ANALOG_OUTPUT for pin: P8_08
   P8_08 as INPUT? pass
   P8_09 as OUTPUT? pass
   P8_09 as INPUT_PULLUP? pass
BeagleBone does not allow ANALOG_OUTPUT for pin: P8_09
   P8_09 as INPUT? pass
   P8_10 as OUTPUT? pass
   P8_10 as INPUT_PULLUP? pass
BeagleBone does not allow ANALOG_OUTPUT for pin: P8_10
   P8_10 as INPUT? pass
   P8_13 as OUTPUT? pass
   P8_13 as INPUT_PULLUP? pass
Error reading directory /sys/devices/platform/ocp/48304000.epwmss/48304200.ehrpwm/pwm
   P8_13 as INPUT? pass
   P8_19 as OUTPUT? pass
   P8_19 as INPUT_PULLUP? pass
Error reading directory /sys/devices/platform/ocp/48304000.epwmss/48304200.ehrpwm/pwm
   P8_19 as INPUT? pass
   P9_11 as OUTPUT? pass
   P9_11 as INPUT_PULLUP? pass
BeagleBone does not allow ANALOG_OUTPUT for pin: P9_11
   P9_11 as INPUT? pass
   P9_13 as OUTPUT? pass
   P9_13 as INPUT_PULLUP? pass
BeagleBone does not allow ANALOG_OUTPUT for pin: P9_13
   P9_13 as INPUT? pass
   P9_14 as OUTPUT? pass
   P9_14 as INPUT_PULLUP? pass
Error reading directory /sys/devices/platform/ocp/48302000.epwmss/48302200.ehrpwm/pwm
   P9_14 as INPUT? pass
   P9_15 as OUTPUT? pass
   P9_15 as INPUT_PULLUP? pass
BeagleBone does not allow ANALOG_OUTPUT for pin: P9_15
   P9_15 as INPUT? pass
   P9_16 as OUTPUT? pass
   P9_16 as INPUT_PULLUP? pass
Error reading directory /sys/devices/platform/ocp/48302000.epwmss/48302200.ehrpwm/pwm
   P9_16 as INPUT? pass
   P9_17 as OUTPUT? pass
   P9_17 as INPUT_PULLUP? pass
BeagleBone does not allow ANALOG_OUTPUT for pin: P9_17
   P9_17 as INPUT? pass
   P9_18 as OUTPUT? pass
   P9_18 as INPUT_PULLUP? pass
BeagleBone does not allow ANALOG_OUTPUT for pin: P9_18
   P9_18 as INPUT? pass
   P9_21 as OUTPUT? pass
   P9_21 as INPUT_PULLUP? pass
Error reading directory /sys/devices/platform/ocp/48300000.epwmss/48300200.ehrpwm/pwm
   P9_21 as INPUT? pass
   P9_22 as OUTPUT? pass
   P9_22 as INPUT_PULLUP? pass
Error reading directory /sys/devices/platform/ocp/48300000.epwmss/48300200.ehrpwm/pwm
   P9_22 as INPUT? pass
   P9_23 as OUTPUT? pass
   P9_23 as INPUT_PULLUP? pass
BeagleBone does not allow ANALOG_OUTPUT for pin: P9_23
   P9_23 as INPUT? pass
   P9_24 as OUTPUT? pass
   P9_24 as INPUT_PULLUP? pass
BeagleBone does not allow ANALOG_OUTPUT for pin: P9_24
   P9_24 as INPUT? pass
   P9_25 as OUTPUT? pass
   P9_25 as INPUT_PULLUP? pass
BeagleBone does not allow ANALOG_OUTPUT for pin: P9_25
   P9_25 as INPUT? pass
   P9_26 as OUTPUT? pass
   P9_26 as INPUT_PULLUP? pass
BeagleBone does not allow ANALOG_OUTPUT for pin: P9_26
   P9_26 as INPUT? pass
   P9_27 as OUTPUT? pass
   P9_27 as INPUT_PULLUP? pass
BeagleBone does not allow ANALOG_OUTPUT for pin: P9_27
   P9_27 as INPUT? pass
   P9_28 as OUTPUT? pass
   P9_28 as INPUT_PULLUP? pass
   P9_28 as INPUT? pass
   P9_29 as OUTPUT? pass
   P9_29 as INPUT_PULLUP? pass
Error reading directory /sys/devices/platform/ocp/48300000.epwmss/48300200.ehrpwm/pwm
   P9_29 as INPUT? pass
BeagleBone does not allow ANALOG_OUTPUT for pin: P9_30
   P9_31 as OUTPUT? pass
   P9_31 as INPUT_PULLUP? pass
Error reading directory /sys/devices/platform/ocp/48300000.epwmss/48300200.ehrpwm/pwm
   P9_31 as INPUT? pass
BeagleBone does not allow ANALOG_OUTPUT for pin: P9_33
BeagleBone does not allow ANALOG_OUTPUT for pin: P9_35
BeagleBone does not allow ANALOG_OUTPUT for pin: P9_36
BeagleBone does not allow ANALOG_OUTPUT for pin: P9_37
BeagleBone does not allow ANALOG_OUTPUT for pin: P9_38
BeagleBone does not allow ANALOG_OUTPUT for pin: P9_39
BeagleBone does not allow ANALOG_OUTPUT for pin: P9_40
   P9_41 as OUTPUT? pass
   P9_41 as INPUT_PULLUP? pass
BeagleBone does not allow ANALOG_OUTPUT for pin: P9_41
   P9_41 as INPUT? pass
   P9_42 as OUTPUT? pass
   P9_42 as INPUT_PULLUP? pass
   P9_42 as INPUT? pass
```

I get the following JSON results:

```json
[{"pin":"USR0","modes":["OUTPUT"]},{"pin":"USR1","modes":["OUTPUT"]},{"pin":"USR2","modes":["OUTPUT"]},{"pin":"USR3","modes":["OUTPUT"]},{"pin":"P8_07","modes":["OUTPUT","INPUT_PULLUP","INPUT"]},{"pin":"P8_08","modes":["OUTPUT","INPUT_PULLUP","INPUT"]},{"pin":"P8_09","modes":["OUTPUT","INPUT_PULLUP","INPUT"]},{"pin":"P8_10","modes":["OUTPUT","INPUT_PULLUP","INPUT"]},{"pin":"P8_13","modes":["OUTPUT","INPUT_PULLUP","INPUT"]},{"pin":"P8_19","modes":["OUTPUT","INPUT_PULLUP","INPUT"]},{"pin":"P9_11","modes":["OUTPUT","INPUT_PULLUP","INPUT"]},{"pin":"P9_13","modes":["OUTPUT","INPUT_PULLUP","INPUT"]},{"pin":"P9_14","modes":["OUTPUT","INPUT_PULLUP","INPUT"]},{"pin":"P9_15","modes":["OUTPUT","INPUT_PULLUP","INPUT"]},{"pin":"P9_16","modes":["OUTPUT","INPUT_PULLUP","INPUT"]},{"pin":"P9_17","modes":["OUTPUT","INPUT_PULLUP","INPUT"]},{"pin":"P9_18","modes":["OUTPUT","INPUT_PULLUP","INPUT"]},{"pin":"P9_21","modes":["OUTPUT","INPUT_PULLUP","INPUT"]},{"pin":"P9_22","modes":["OUTPUT","INPUT_PULLUP","INPUT"]},{"pin":"P9_23","modes":["OUTPUT","INPUT_PULLUP","INPUT"]},{"pin":"P9_24","modes":["OUTPUT","INPUT_PULLUP","INPUT"]},{"pin":"P9_25","modes":["OUTPUT","INPUT_PULLUP","INPUT"]},{"pin":"P9_26","modes":["OUTPUT","INPUT_PULLUP","INPUT"]},{"pin":"P9_27","modes":["OUTPUT","INPUT_PULLUP","INPUT"]},{"pin":"P9_28","modes":["OUTPUT","INPUT_PULLUP","INPUT"]},{"pin":"P9_29","modes":["OUTPUT","INPUT_PULLUP","INPUT"]},{"pin":"P9_31","modes":["OUTPUT","INPUT_PULLUP","INPUT"]},{"pin":"P9_41","modes":["OUTPUT","INPUT_PULLUP","INPUT"]},{"pin":"P9_42","modes":["OUTPUT","INPUT_PULLUP","INPUT"]}]
```

troywweber7, 2017-08-07:
Furthermore when I `$ ls /sys/devices/platform/ocp/*.epwmss/` I get:

```bash
$ ls /sys/devices/platform/ocp/*.epwmss/
/sys/devices/platform/ocp/48300000.epwmss/:
48300100.ecap  48300200.pwm  driver_override  of_node  subsystem
48300180.eqep  driver        modalias         power    uevent

/sys/devices/platform/ocp/48302000.epwmss/:
48302180.eqep  48302200.pwm  driver  driver_override  modalias  of_node  power  subsystem  uevent

/sys/devices/platform/ocp/48304000.epwmss/:
48304100.ecap  48304200.pwm  driver_override  of_node  subsystem
48304180.eqep  driver        modalias         power    uevent
```

troywweber7, 2017-08-07:
I tried adding more cape overlays to no avail:

```bash
$ grep -o 'INPUT_' results_pinMode.json | wc -l
25
$ cat $SLOTS
 0: PF----  -1 
 1: PF----  -1 
 2: PF----  -1 
 3: PF----  -1 
 4: P-O-L-   0 Override Board Name,00A0,Override Manuf,cape-universaln
 6: P-O-L-   1 Override Board Name,00A0,Override Manuf,BB-ADC
 7: P-O-L-   2 Override Board Name,00A0,Override Manuf,BB-PWM0
 8: P-O-L-   3 Override Board Name,00A0,Override Manuf,BB-PWM1
 9: P-O-L-   4 Override Board Name,00A0,Override Manuf,BB-PWM2
```

troywweber7, 2017-08-07:
I rebooted and tried with univ-bbgw overlay and was able to get back control of my 42 pins back. Although, when removing the cape, the kernel panics and you have to hard reboot. And then its back to 25.

adityapatadia, 2017-12-24:
This library is abandoned. Please check notice in README for further information.

closed 2017-12-24 by adityapatadia