theoctal/octalbonescript #79: cannot set PWM pin to ANALOG_OUTPUT

troywweber7, 2017-08-07:
When I try to set pin `P8_13`, I get an error and I think maybe it is looking in the wrong directory. I'm on a beaglebone green wireless with Debian 8.9 (jessie) and all updates installed. Within node interpreter:

```
obs.pinModeSync('P8_13', obs.ANALOG_OUTPUT)
Error reading directory /sys/devices/platform/ocp/48304000.epwmss/48304200.ehrpwm/pwm
TypeError: Cannot read property 'message' of undefined
  at Object.f.pinModeSync (/home/deploy/code/dicegame/node_modules/octalbonescript/index.js:326:27)
  at repl:1:5
  at ContextifyScript.Script.runInThisContext (vm.js:23:33)
  at REPLServer.defaultEval (repl.js:339:29)
  at bound (domain.js:280:14)
  at REPLServer.runBound [as eval] (domain.js:293:12)
  at REPLServer.onLine (repl.js:536:10)
  at emitOne (events.js:101:20)
  at REPLServer.emit (events.js:191:7)
  at REPLServer.Interface._onLine (readline.js:241:10)
  at REPLServer.Interface._line (readline.js:590:8)
  at REPLServer.Interface._ttyWrite (readline.js:869:14)
  at REPLServer.self._ttyWrite (repl.js:609:7)
  at ReadStream.onkeypress (readline.js:120:10)
  at emitTwo (events.js:106:13)
  at ReadStream.emit (events.js:194:7)
  at emitKeys (internal/readline.js:402:14)
  at emitKeys.next (<anonymous>:null:null)
  at ReadStream.onData (readline.js:980:36)
  at emitOne (events.js:96:13)
  at ReadStream.emit (events.js:191:7)
  at readableAddChunk (_stream_readable.js:178:18)
  at ReadStream.Readable.push (_stream_readable.js:136:10)
  at TTY.onread (net.js:561:20
```

When I investigate that directory, I don't see the directory it is looking for:

```bash
$ ls /sys/devices/platform/ocp/48304000.epwmss/48304200.ehrpwm
ls: cannot access /sys/devices/platform/ocp/48304000.epwmss/48304200.ehrpwm: No such file or directory
```

troywweber7, 2017-08-07:
Digging in the code from which this project is forked I find this: https://github.com/jadonk/bonescript/blob/1836388c3f3e770899ddafd861556155a65d39f7/src/hw_mainline.js#L97.

This indicates to me that octalbonescript hasn't yet pulled in that change to account for both `.pwm` and `.ehrpwm`.

adityapatadia, 2017-12-24:
This library is abandoned. Please check notice in README for further information.

closed 2017-12-24 by adityapatadia