theoctal/octalbonescript #81: How can I use GPIO pins and UART4?

maxwellhadley, 2017-09-25:
I'm using a BBB, with an 18 month old OS:

    uname - a
    Linux redbox 4.1.18-ti-r53 #1 SMP Thu Mar 17 19:41:05 UTC 2016 armv7l GNU/Linux
    node --version
   v6.11.3

I have modified /etc/default/capemgr to load BBB-UART4 so I can use /dev/ttyO4 from serialport:

    CAPE=BB-UART4

I'm using octalbonescript from Node-RED & the supplied Beaglebone nodes. If I set the environment variable `AUTO_LOAD_CAPE=0` then I can use /dev/ttyO4, but all GPIO pins are 'not available':

    VError: The pin P9_42 is not availble to write. Please make sure it is not used by another cape.

If I run without the environment variable, then I get the 'file already exists' error #65. If I remove the line from /etc/default/capemgr that loads BB-UART4 and run without the environment variable, I can access the pins but not /dev/ttyO4.

How can I access both GPIO pins and UART4 from the same application?

maxwellhadley, 2017-09-27:
I found a workaround. First, install an up-to-date bb-cape-overlays:

    sudo apt-get install bb-cape-overlays

Then, change /etc/default/capemgr to load the appropriate universal cape:

    CAPE=cape-universaln

After rebooting, you should see the device file for UART4, `/dev/ttyO4`, is now present. Next, use the `config-pin` script (which should have been installed by `bb-cape-overlays`) to set the pinmux to route the UART TxD & RxD to the appropriate pins:

    sudo config-pin P9_11 uart
    sudo config-pin P9_13 uart

(If this step is missing, you can happily open `/dev/ttyO4` and use it without errors, but the signals never get off the chip - it took me hours to work this bit out!)

Finally, I run my Node application (Node-RED) startup script with the appropriate environment variable set to let `octalbonescript` know a universal cape is already loaded. This is necessary because you have to load the universal cape *before* you can use `config-pin` to control the pinmux:

    AUTO_LOAD_CAPE=0 node-red-pi

This script starts up Node-RED in low-memory mode, which in turn loads `octalbonescript` to access GPIO pins, and `serialport` to access UART4. Which it does, successfully. I have added my user account to the `gpio` group so I don't need to be root to access the GPIO pins

adityapatadia, 2017-12-24:
This library is abandoned. Please check notice in README for further information.

closed 2017-12-24 by adityapatadia